Ladder99/fanuc-driver #1: Not an Issue

thingevents, 2021-06-08:
Hi, 

First off this is awesome work, really like the approach you are taking, would be great to extend this to toher cnc connectrollers, let me know if interested!

Could you help me, I was wondering how to implement alarms, but get caught up on the Collector implementation, I can see the alarms are read from the Fanuc machine but I dont know how to implement the Apply and Peel Vaneers? Be great to get this working. thanks.

MRIIOT, 2021-06-09:
@thingevents Thanks.  The thought is that the concepts in the base-driver repository can be extended to other hardware.  What other hardware were you thinking of?  I see there are open source libs available for Rockwell, Opto22, S7.  But those are more of a read a tag and report it, but could still benefit from packaging multiple tags into a payload that has more context and structure depending on the machine, process, and target system.

@tukusejssirs What data are you interested in collecting?

tukusejssirs, 2021-06-09:
@MRIIOT, right now we want to collect:
- production data (program name, pieces produced, total number of pieces to be produced, start timestamp, end timestamp);
- alarm codes and messages.

In the future we might want to expand these with other data (like axes position and RPM).

All of that on the controllers listed in [this comment](https://github.com/Ladder99/fanuc-driver/issues/1#issuecomment-856568233).

MRIIOT, 2021-06-09:
@tukusejssirs 

How do you differentiate: pieces produced from total number of pieces to be produced?
How do you define: start and end timestamp?

tukusejssirs, 2021-06-09:
> How do you differentiate: pieces produced from total number of pieces to be produced?
> How do you define: start and end timestamp?

Well, _total number of pieces_ is a number of pieces set by operator to be produced using a particular program. _Pieces produced_ are pieces that are already finished.

MRIIOT, 2021-06-10:
I'll create something to pull 6713 and 6711 params, and the rest of the stuff.

[image]

MRIIOT, 2021-06-10:
@thingevents 

Try [this](https://github.com/Ladder99/fanuc-driver/blob/main/fanuc/collectors/UseCase01.cs).  I have not tried them yet, so let's see if you get valid data.

tukusejssirs, 2021-06-10:
> I'll create something to pull 6713 and 6711 params, and the rest of the stuff.

Thanks. It might be a good idea to get all three values. To my understanding (checked this by operators), these are the meanings of those values:
- `6711` is parts count produced since a particular program start (or since last clearing of this value?);
- `6712` total parts count produced on the machines in its lifetime (or since the last OS installation?);
- `6713` is total parts count to be produced (set by operator).

---

Anyway, even other machine state information would be useful to use, like if the machine runs in automatic/manual mode or whatever else is available.

closed 2022-02-01 by MRIIOT