Ladder99/fanuc-driver #2: Is it possible to run this directly on a Linux server?

tukusejssirs, 2021-06-08:
I don’t code in C/C++/C# (I can more or less read the code and compile it using `gcc`) and have never really used Docker images, so treat me as a newbie in these areas.

---

I’d like to run `fanuc-driver` directly on a Linux (CentOS 8) server. Is it possible? Could you post me some instructions to make this happen?

All I actually need is to output the acquired data (especially production, alarms, status) ideally in JSON format to be processed/used outside of `fanuc-driver`.

MRIIOT, 2021-06-09:
It should be.  What bitness of CentOS are you running, 32bit or 64bit?  Also what architecture, x86, arm, ...?  We will definitely have to modify the [pinvoke](https://github.com/Ladder99/fanuc-driver/blob/ced59ac2fc7f31b3e736367fddd6b86516e42418/fanuc/fwlib32.cs#L7979) signatures for each method you need in order to call into the correct Focas Linux library.

The nice thing about Docker is that even if I'm running ARM 64 bit OS, for which there is no Focas library, I can still run on top of a 32 bit image and be OK.... Kinda like a virtual machine.

1. Install .NET Core 3.1
2. Clone repo
3. Modify config.yml
4. Modify nlog.config
5. build and run

What failure do you get ?

```
export DOTNET_ROOT=$HOME/dotnet
export PATH=$PATH:$HOME/dotnet

cd ~

git clone --recurse-submodules -j8 https://github.com/Ladder99/fanuc-driver.git  

cd fanuc-driver/fanuc  

dotnet build  /nowarn:CS0618 /nowarn:CS8632 /nowarn:CS1998 

./bin/Debug/netcoreapp3.1/fanuc

tukusejssirs, 2021-06-09:
> It should be. What bitness of CentOS are you running, 32bit or 64bit? Also what architecture, x86, arm, ...?

x86_64

> We will definitely have to modify the [pinvoke](https://github.com/Ladder99/fanuc-driver/blob/ced59ac2fc7f31b3e736367fddd6b86516e42418/fanuc/fwlib32.cs#L7979) signatures for each method you need in order to call into the correct Focas Linux library.

Okay, tell me what you need from me, as I have no idea what a pinvoke is (I’ve read something about them [here](https://docs.microsoft.com/en-us/dotnet/standard/native-interop/pinvoke) though). :wink:

> The nice thing about Docker is that even if I'm running ARM 64 bit OS, for which there is no Focas library, I can still run on top of a 32 bit image and be OK.... Kinda like a virtual machine.

Yeah, I see what the Docker is about, but I’d like to minimise the list of dependencies and the app layers as much as possible and as much as I can.

> What failure do you get ?

See #4.