Ladder99/fanuc-driver #4: Unable to load shared library 'FWLIB32.dll'

tukusejssirs, 2021-06-08:
I have encountered the following error. I am not sure if the error is raised because I haven’t placed a library at the required location. Actually, I didn’t move/copy any library files, because (as I understand it) `fanuc-driver` should contain the required libraries (of course, not for all Fanuc controller versions).

```
2021/06/08 20:04:49.217|ERROR|[m1] Collector initialization failed. System.DllNotFoundException: Unable to load shared library 'FWLIB32.dll' or one of its dependencies. In order to help diagnose loading problems, consider setting the LD_DEBUG environment variable: libFWLIB32.dll: cannot open shared object file: No such file or directory
```

I found `fanuc/Fwlib32.dll`, but it does not even symlinking it to `fanuc/FWLIB32.dll` wouldn’t solve the problem (presuming that the problem is in the case sensitivity of Linux).

MRIIOT, 2021-06-09:
This is on Linux, correct?

I bet you what is happening is that one of the [native calls](https://github.com/Ladder99/fanuc-driver/blob/main/fanuc/fwlib32.cs) being made is not marked with the proper constant for Linux.  I have been focusing on Windows and ARM only so far.

If you are using the default config.yaml then the class to look at would be strategy_type: [l99.driver.fanuc.collectors.Basic07](https://github.com/Ladder99/fanuc-driver/blob/main/fanuc/collectors/Basic07.cs).  It would be easier to modify the config.yaml to call Basic01 collector, which has much less calls:

```yml
strategy_type: l99.driver.fanuc.collectors.Basic01, fanuc
```

tukusejssirs, 2021-06-09:
> This is on Linux, correct?

Yes, on Fedora 32 x86_64, though the production will be on CentOS 8 x86_64.

> I bet you what is happening is that one of the [native calls](https://github.com/Ladder99/fanuc-driver/blob/main/fanuc/fwlib32.cs) being made is not marked with the proper constant for Linux. I have been focusing on Windows and ARM only so far.
> 
> If you are using the default config.yaml then the class to look at would be strategy_type: [l99.driver.fanuc.collectors.Basic07](https://github.com/Ladder99/fanuc-driver/blob/main/fanuc/collectors/Basic07.cs). It would be easier to modify the config.yaml to call Basic01 collector, which has much less calls:
> 
> ```yaml
> strategy_type: l99.driver.fanuc.collectors.Basic01, fanuc
> ```

I tried several strategy types and handler types, but the error is still the same.

Is there a way to help you somehow?

MRIIOT, 2021-06-10:
Basically as I go through the functions I need to call I add a conditional to fwlib32.cs for that method.  So here:

[image]

I change that to:

[image]

let me go through it and try it on my linux box.

MRIIOT, 2021-06-10:
Try the latest push.  I tried on Ubuntu 20.04.

`dotnet build  /nowarn:CS0618 /nowarn:CS8632 /nowarn:CS1998 -p:DefineConstants=LINUX64`

You should get this error.

[image]

Here is the problem: https://stackoverflow.com/questions/52984790/fieldsoffset-in-structs-in-x64-c-sharp-application

We have three options: 

1. go through fwlib32.cs and make it work for 64bit.
2. test on 32bit.
3. run in Docker using a 32bit image.

MRIIOT, 2021-06-10:
Well looks like x86 is not supported.

https://github.com/dotnet/core/blob/main/os-lifecycle-policy.md

So back to #1.  I'm not sure of any other option.

MRIIOT, 2021-06-10:
Also, if you comment out line 65,66 from Basic01, it should run for you.  

[image]

If you want to fiddle around and try to make RdParam call work on 64 bit, that would be appreciated.

[image]

tukusejssirs, 2021-06-10:
> Try the latest push. I tried on Ubuntu 20.04.
>
> `dotnet build /nowarn:CS0618 /nowarn:CS8632 /nowarn:CS1998 -p:DefineConstants=LINUX64`
>
> You should get this error.

I’ve got the same error.

> Also, if you comment out line 65,66 from Basic01, it should run for you.

No, it couldn’t run. Now I’ve got the original error again (`Collector initialization failed. System.DllNotFoundException: Unable to load shared library 'FWLIB32.dll' or one of its dependencies.`).

FYI, currently I test `fanuc-driver` on Doosan Puma 2600LM with Fanuc Series 0i-TF controller.

> We have three options:
>
>     1. go through fwlib32.cs and make it work for 64bit.
>
>     2. test on 32bit.
>
>     3. run in Docker using a 32bit image.

Optioin 1 would be ideal from my point of view, if not, then 32-bit program.

> Well looks like x86 is not supported.
>
> https://github.com/dotnet/core/blob/main/os-lifecycle-policy.md
>
> So back to trying to get it running on 64bit? I'm not sure of any other option.

Well, as I’d like to run `fanuc-driver` on a 64-bit OS, that would be awesome, however, seeing the limitations, is it possible with the resources we (you) have?

> If you want to fiddle around and try to make RdParam call work on 64 bit, that would be appreciated.

I’d like to do that, I just some pointers–what should I fiddle with in the shared code? Change the `FieldOffset`s? Is this info available somewhere in the Fanuc manuals?

---

Anyway, a big thank you for trying and helping me out! I should by you a beer or something. :wink:

MRIIOT, 2021-06-10:
I don't understand why the latest push wouldn't run for you.  

`dotnet build /nowarn:CS0618 /nowarn:CS8632 /nowarn:CS1998 -p:DefineConstants=LINUX64`

You are calling out Basic01.cs in config?

`strategy_type: l99.driver.fanuc.collectors.Basic01, fanuc`

Is this a limitation of which *nix flavor the library can run on?

And last I heard, Linux support has been dropped from Focas because "too many Linux variants".

Yes, sorry, I meant port it to 64bit.  This is more of an exercise in p/invoke and marshaling types between unmanaged code and .NET, like this link (https://stackoverflow.com/questions/52984790/fieldsoffset-in-structs-in-x64-c-sharp-application), but you are not even getting this type of error.

Re Focas1/Focas2: the fwlib32.cs file comes from the A02B-0207-K737 CD.  

[image]

Can I convince you to run it on a Raspberry Pi (bare metal or docker / armv7) or [Nano Pi](https://www.friendlyarm.com/index.php?route=product/product&product_id=284) (docker / aarch64)?

Let's drink a beer after we get this working.

tukusejssirs, 2021-06-10:
> I don't understand why the latest push wouldn't run for you.
> 
> `dotnet build /nowarn:CS0618 /nowarn:CS8632 /nowarn:CS1998 -p:DefineConstants=LINUX64`
> 
> You are calling out Basic01.cs in config?
> 
> `strategy_type: l99.driver.fanuc.collectors.Basic01, fanuc`

Definitely yes. Below is the content of `config.yaml` I used.

<details>
   <summary><code>config.yaml</code></summary>

```yaml
---
machines:
  - id: test
    enabled: !!bool true
    type: l99.driver.fanuc.FanucMachine, fanuc
    strategy_type: l99.driver.fanuc.collectors.Basic01, fanuc
    handler_type: l99.driver.fanuc.handlers.Native, fanuc
    sweep_ms: !!int 1000
    net_ip: 192.168.1.1
    net_port: !!int 8193
    net_timeout_s: !!int 10
    broker:
      enabled: !!bool false
      net_ip: 10.20.30.102
      net_port: !!int 1883
      auto_connect: !!bool false
      publish_status: !!bool true
      publish_arrivals: !!bool true
      publish_changes: !!bool true
      publish_disco: !!bool true
      disco_base_topic: fanuc
...
```

</details>

> Is this a limitation of which \*nix flavor the library can run on?

I can’t answer that.

> And last I heard, Linux support has been dropped from Focas because "too many Linux variants".

Yes, so I heard too. However, I personally don’t really care if they support it as long as it works. :wink:

> Yes, sorry, I meant port it to 64bit. This is more of an exercise in p/invoke and marshaling types between unmanaged code and .NET, like this link (https://stackoverflow.com/questions/52984790/fieldsoffset-in-structs-in-x64-c-sharp-application), but you are not even getting this type of error.

I see. I’ll do my best, but keep in mind that my C/C++/C# knowledge is minimal at best.

> Re Focas1/Focas2: the fwlib32.cs file comes from the A02B-0207-K737 CD.

There is only one mention of FOCAS2 in that screenshot: `Since the functions of all FOCAS2 is not defined, add the function not defined in this file.` Does it mean that all functions that are added to FOCAS2 in comparison with FOCAS1 are not defined in `fwlib32.cs`/`fwlib32.vb` (and therefore `fwlib32.cs`/`fwlib32.vb` is not meant to be working with FOCAS2) or the only added function is `not defined` (as in one function named `not_defined()`), and therefore all that is needed to make `fwlib32.cs`/`fwlib32.vb` compatible with FOCAS2 is to defined that function?

> Can I convince you to run it on a Raspberry Pi (bare metal or docker / armv7) or [Nano Pi](https://www.friendlyarm.com/index.php?route=product/product&product_id=284) (docker / aarch64)?

Not really. I’d rather run a Docker image that having an extra hardware to a dedicated bare-metal/physical server + VMware ESXi + VM (CentOS 8)

> Let's drink a beer after we get this working.

Agreed! :+1:

MRIIOT, 2021-06-10:
[This](https://github.com/dlozina/measuring-machine-db/blob/e4d884788407ecd7a8c66bdc9f1571c42639aaa2/FOCAS2Library/fwlib64.cs) should help with porting to 64bit.

But again, the Basic01.cs example runs fine for me on Ubuntu.  But not for you...

tukusejssirs, 2021-06-10:
> [This](https://github.com/dlozina/measuring-machine-db/blob/e4d884788407ecd7a8c66bdc9f1571c42639aaa2/FOCAS2Library/fwlib64.cs) should help with porting to 64bit.

Thanks, I’ll look at it.

> But again, the Basic01.cs example runs fine for me on Ubuntu. But not for you...

Are there any additional requirements/dependencies that are fullfilled on Ubuntu by default, but not on Fedora / RHEL-like systems?

MRIIOT, 2021-06-10:
I don't know.  Do you want to spin up a Ubuntu VM and we can try there?

> Does it mean that all functions that are added to FOCAS2 in comparison with FOCAS1 are not defined in fwlib32.cs/fwlib32.vb (and therefore fwlib32.cs/fwlib32.vb is not meant to be working with [...]

Maybe ?

tukusejssirs, 2021-06-10:
> I don't know. Do you want to spin up a Ubuntu VM and we can try there?

I’ll try doing that.

> > Does it mean that **(1) all functions that are added to FOCAS2 in comparison with FOCAS1 are not defined in fwlib32.cs/fwlib32.vb (and therefore fwlib32.cs/fwlib32.vb is not meant to be working with FOCAS2) or (2) the only added function is not defined (as in one function named not_defined()), and therefore all that is needed to make fwlib32.cs/fwlib32.vb compatible with FOCAS2 is to defined that function?**
> 
> Maybe ?

_Which_ of the two possible explations are _maybe_ valid? :wink:

MRIIOT, 2021-06-10:
2 !

MRIIOT, 2021-06-10:
@tukusejssirs I sent you a message on telegram.

tukusejssirs, 2021-06-10:
Probably a stupid question: can’t we use 32-bit library in 64-bit C# code on x86_64 multilib Linux?

MRIIOT, 2021-06-10:
I don't think you can mix bitness in .NET process...

tukusejssirs, 2021-06-10:
Then for running `fanuc-driver` without VM/Docker/RPi, I see only two other possibilities:
- build a 32-bit version of `fanuc-driver`:
  - I have no idea if this would cause some noticable performance degradation;
- port 32-bit `fwlib` to 64-bit:
  - do we actually know that making RdParam call work on 64 bit would solve this? are their any other stuff that are missing from the x64 version of `fwlib`?

MRIIOT, 2021-06-10:
32 bit runs great on a Raspberry Pi.

We don't know.  And we don't know what other issues will come up for 64bit.  I would assume that fwlib64.dll for Windows would be better tested than the Linux 64bit shared objects.

tukusejssirs, 2021-06-10:
I compared your [`fwlib32.cs`](https://github.com/Ladder99/fanuc-driver/blob/main/fanuc/fwlib32.cs) with dlozina’s [`fwlib64.cs`](https://github.com/dlozina/measuring-machine-db/blob/e4d884788407ecd7a8c66bdc9f1571c42639aaa2/FOCAS2Library/fwlib64.cs) using `git diff --no-index`. Of course, I pre-processed the two files in order to remove comments, most extra spaces and replacing `FWLIB32.dll` with `FWLIB64.dll`. You can find the compared files here → [`fwlib32_orig.txt`](https://github.com/Ladder99/fanuc-driver/files/6632574/fwlib32_orig.txt), [`fwlib64_mdfd_dlozina.txt`](https://github.com/Ladder99/fanuc-driver/files/6632577/fwlib64_mdfd_dlozina.txt). This produces 1587 lines of `diff` code vs ~12,000 lines in each files. Here’s the diff → [fwlib32_64_diff.txt](https://github.com/Ladder99/fanuc-driver/files/6632630/fwlib32_64_diff.txt).

As I am not a C/C++/C# coder, I believe you could better see the changes required to improve the x64 support in `fwlib32.cs`. From what I have seen, he has some additional/different classes and also defines some additional variables.

MRIIOT, 2021-06-10:
Looking at that diff, I'm not sure how 64bit compatible that specific fwlib64.cs really is.

From my understanding the changes would be something like this:

```c#
32bit:

[StructLayout(LayoutKind.Explicit)]
public class IODBPSD_3
{
    [FieldOffset(0)]
    public short datano;

    [FieldOffset(2)]
    public short type;

    [FieldOffset(4), MarshalAs(UnmanagedType.ByValArray, SizeConst = MAX_AXIS)]
    public byte[] cdatas = new byte[MAX_AXIS];

    [FieldOffset(4), MarshalAs(UnmanagedType.ByValArray, SizeConst = MAX_AXIS)]
    public short[] idatas = new short[MAX_AXIS];

    [FieldOffset(4), MarshalAs(UnmanagedType.ByValArray, SizeConst = MAX_AXIS)]
    public int[] ldatas = new int[MAX_AXIS];
}

64bit:

[StructLayout(LayoutKind.Sequential)]
public class IODBPSD_3
{
    public short datano;

    public short type;

    public IODBPSD_3_INNER inner;
}

[StructLayout(LayoutKind.Explicit)]
public class IODBPSD_3_INNER
{
    [FieldOffset(0), MarshalAs(UnmanagedType.ByValArray, SizeConst = MAX_AXIS)]
    public byte[] cdatas = new byte[MAX_AXIS];

    [FieldOffset(0), MarshalAs(UnmanagedType.ByValArray, SizeConst = MAX_AXIS)]
    public short[] idatas = new short[MAX_AXIS];

    [FieldOffset(0), MarshalAs(UnmanagedType.ByValArray, SizeConst = MAX_AXIS)]
    public int[] ldatas = new int[MAX_AXIS];
}
```

tukusejssirs, 2021-06-10:
I see. Would it be possible to use `fwlib64.cs` without (much) modifications in `fanuc-driver`? I could test it it would work like that, but could you create a simple test script (or a test branch) for me that would utilise `fwlib64.cs`? What I mean is when running on x86_64 OS, simply use `fwlib64.cs`.

tukusejssirs, 2021-06-10:
When testing the commented out lines in `Basic01.cs`, I have forgotten to recompile the changes. :facepalm: Now I get the same error as if I haven’t commented them out (`Could not load type 'IODBPSD_3' from assembly 'fanuc, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null' because it contains an object field at offset 4 that is incorrectly aligned or overlapped by a non-object field.`).

---

I tested this on Ubuntu 20.04.01 in VirtualBox. I installed `dotnet` v3.1 using [these instructions](https://docs.microsoft.com/en-us/dotnet/core/install/linux-ubuntu#1910-) (did you install them using `snap`?). Then I did as follows:

```bash
git clone --recurse-submodules git@github.com:Ladder99/fanuc-driver.git
DOTNET_CLI_TELEMETRY_OPTOUT=true dotnet build /nowarn:CS0618 /nowarn:CS8632 /nowarn:CS1998 -p:DefineConstants=LINUX64 fanuc-driver/fanuc/fanuc.csproj
cd fanuc-driver/fanuc
DOTNET_CLI_TELEMETRY_OPTOUT=true dotnet run /nowarn:CS0618 /nowarn:CS8632 /nowarn:CS1998 fanuc.csproj
```

Some warnings were raised, see below the output of the `dotnet` commands.

<details>
  <summary><code>dotnet build</code></summary>

```bash
Microsoft (R) Build Engine version 16.7.1+52cd83677 for .NET
Copyright (C) Microsoft Corporation. All rights reserved.

  Determining projects to restore...
  Restored /home/ts/fanuc-driver/fanuc/fanuc.csproj (in 1.16 min).
collectors/UseCase01.cs(15,13): warning CS4014: Because this call is not awaited, execution of the current method continues before the call is completed. Consider applying the 'await' operator to the result of the call. [/home/ts/fanuc-driver/fanuc/fanuc.csproj]
collectors/UseCase01.cs(17,13): warning CS4014: Because this call is not awaited, execution of the current method continues before the call is completed. Consider applying the 'await' operator to the result of the call. [/home/ts/fanuc-driver/fanuc/fanuc.csproj]
collectors/UseCase01.cs(19,13): warning CS4014: Because this call is not awaited, execution of the current method continues before the call is completed. Consider applying the 'await' operator to the result of the call. [/home/ts/fanuc-driver/fanuc/fanuc.csproj]
collectors/Basic08.cs(15,13): warning CS4014: Because this call is not awaited, execution of the current method continues before the call is completed. Consider applying the 'await' operator to the result of the call. [/home/ts/fanuc-driver/fanuc/fanuc.csproj]
collectors/Basic08.cs(17,13): warning CS4014: Because this call is not awaited, execution of the current method continues before the call is completed. Consider applying the 'await' operator to the result of the call. [/home/ts/fanuc-driver/fanuc/fanuc.csproj]
collectors/Basic08.cs(22,13): warning CS4014: Because this call is not awaited, execution of the current method continues before the call is completed. Consider applying the 'await' operator to the result of the call. [/home/ts/fanuc-driver/fanuc/fanuc.csproj]
collectors/Basic08.cs(24,13): warning CS4014: Because this call is not awaited, execution of the current method continues before the call is completed. Consider applying the 'await' operator to the result of the call. [/home/ts/fanuc-driver/fanuc/fanuc.csproj]
collectors/Basic08.cs(26,13): warning CS4014: Because this call is not awaited, execution of the current method continues before the call is completed. Consider applying the 'await' operator to the result of the call. [/home/ts/fanuc-driver/fanuc/fanuc.csproj]
collectors/Basic08.cs(28,13): warning CS4014: Because this call is not awaited, execution of the current method continues before the call is completed. Consider applying the 'await' operator to the result of the call. [/home/ts/fanuc-driver/fanuc/fanuc.csproj]
collectors/Basic08.cs(33,13): warning CS4014: Because this call is not awaited, execution of the current method continues before the call is completed. Consider applying the 'await' operator to the result of the call. [/home/ts/fanuc-driver/fanuc/fanuc.csproj]
collectors/Basic08.cs(35,13): warning CS4014: Because this call is not awaited, execution of the current method continues before the call is completed. Consider applying the 'await' operator to the result of the call. [/home/ts/fanuc-driver/fanuc/fanuc.csproj]
collectors/FanucCollector2.cs(250,29): warning CS4014: Because this call is not awaited, execution of the current method continues before the call is completed. Consider applying the 'await' operator to the result of the call. [/home/ts/fanuc-driver/fanuc/fanuc.csproj]
collectors/FanucCollector2.cs(311,25): warning CS4014: Because this call is not awaited, execution of the current method continues before the call is completed. Consider applying the 'await' operator to the result of the call. [/home/ts/fanuc-driver/fanuc/fanuc.csproj]
collectors/FanucCollector2.cs(335,29): warning CS4014: Because this call is not awaited, execution of the current method continues before the call is completed. Consider applying the 'await' operator to the result of the call. [/home/ts/fanuc-driver/fanuc/fanuc.csproj]
collectors/FanucCollector2.cs(353,29): warning CS4014: Because this call is not awaited, execution of the current method continues before the call is completed. Consider applying the 'await' operator to the result of the call. [/home/ts/fanuc-driver/fanuc/fanuc.csproj]
  fanuc -> /home/ts/fanuc-driver/fanuc/bin/Debug/netcoreapp3.1/fanuc.dll

Build succeeded.

[... cut ...]
   at l99.driver.fanuc.Platform.Connect() in /home/ts/fanuc-driver/fanuc/platform/Connect.cs:line 15
   at l99.driver.fanuc.Platform.ConnectAsync() in /home/ts/fanuc-driver/fanuc/platform/Connect.cs:line 10
   at CallSite.Target(Closure , CallSite , Object )
   at System.Dynamic.UpdateDelegates.UpdateAndExecute1[T0,TRet](CallSite site, T0 arg0)
   at l99.driver.fanuc.collectors.Basic01.CollectAsync() in /home/ts/fanuc-driver/fanuc/collectors/Basic01.cs:line 54|l99.driver.fanuc.collectors.Basic01|
```

tukusejssirs, 2021-06-10:
> Can I convince you to run it on a Raspberry Pi (bare metal or docker / armv7) or [Nano Pi](https://www.friendlyarm.com/index.php?route=product/product&product_id=284) (docker / aarch64)?

On second thoughts: could you send me some commands how could I test out the Docker armv7 version?

That said, I still would like to see a fully functioning native x86_64 Linux version.

MRIIOT, 2021-06-10:
> On second thoughts: could you send me some commands how could I test out the Docker armv7 version?
>
> That said, I still would like to see a fully functioning native x86_64 Linux version.

lol...

[bare metal](https://github.com/Ladder99/fanuc-driver#armv7)

[docker build](https://github.com/Ladder99/fanuc-driver#docker)

[docker compose](https://github.com/Ladder99/fanuc-driver#run-containers-using-docker-compose)

tukusejssirs, 2021-06-10:
I tried to create the Docker image, but it fails.

```bash
sudo dnf -y upgrade
sudo dnf -y install curl docker python3-pip libffi-devel openssl-devel python3-devel python3 python3-pip
sudo systemctl enable --now docker
sudo pip3 install docker-compose
sudo sync
git clone --recurse-submodules -j8 https://github.com/Ladder99/fanuc-driver.git  
cd fanuc-driver && sudo docker build -f Dockerfile.ARM --tag=ladder99/fanuc-driver:latest .
sudo docker push ladder99/fanuc-driver
sudo docker run -it ladder99/fanuc-driver:latest
```

The following command was not successful; see the logs. I haven’t found the `standard_init_linux.go` file in `fanuc-driver` repository nor in the content of any files in it.

<details>
  <summary><code>sudo docker build -f Dockerfile.ARM --tag=ladder99/fanuc-driver:latest .</code></summary>

```bash
Sending build context to Docker daemon  95.68MB
Step 1/18 : FROM mcr.microsoft.com/dotnet/core/runtime:3.1-buster-slim-arm32v7 AS base
3.1-buster-slim-arm32v7: Pulling from dotnet/core/runtime
3035777cd90a: Pull complete 
f53ddbc43e54: Pull complete 
020da15aba73: Pull complete 
bbf3cc160b3d: Pull complete 
Digest: sha256:0bdc1730bc8c570cf0296d1e0544f160901c0579dcb56988c1c6b92d2a7d5259
Status: Downloaded newer image for mcr.microsoft.com/dotnet/core/runtime:3.1-buster-slim-arm32v7
 ---> 66ae5df6d80f
Step 2/18 : WORKDIR /app
 ---> [Warning] The requested image's platform (linux/arm) does not match the detected host platform (linux/amd64) and no specific platform was requested
 ---> Running in ac4a8a6a9c66
Removing intermediate container ac4a8a6a9c66
 ---> 8ccbad70d8c1
Step 3/18 : FROM mcr.microsoft.com/dotnet/core/sdk:3.1-buster-arm32v7 AS build
3.1-buster-arm32v7: Pulling from dotnet/core/sdk
89475607b1df: Pull complete 
88447de5eb3e: Pull complete 
b25b14665fd0: Pull complete 
ba65a331625e: Pull complete 
69595e957c50: Pull complete 
c149b4dcd454: Pull complete 
9266c543d608: Pull complete 
Digest: sha256:e90357a51ef144b46faffab7e119f5bf1c86a2e94aeffb548b25aa5882e1169d
Status: Downloaded newer image for mcr.microsoft.com/dotnet/core/sdk:3.1-buster-arm32v7
 ---> 31f08619a71c
Step 4/18 : WORKDIR /src
 ---> [Warning] The requested image's platform (linux/arm) does not match the detected host platform (linux/amd64) and no specific platform was requested
 ---> Running in 7fc4e6a5291e
Removing intermediate container 7fc4e6a5291e
 ---> df1657298dde
Step 5/18 : COPY ["fanuc/fanuc.csproj", "fanuc/"]
 ---> c8314cda12c1
Step 6/18 : RUN dotnet restore "fanuc/fanuc.csproj"
 ---> [Warning] The requested image's platform (linux/arm) does not match the detected host platform (linux/amd64) and no specific platform was requested
 ---> Running in 9847721d6316
standard_init_linux.go:228: exec user process caused: exec format error
The command '/bin/sh -c dotnet restore "fanuc/fanuc.csproj"' returned a non-zero code: 1
```

</details>

MRIIOT, 2021-06-10:
Aah, you're trying to run an ARMv7 image on x86 architecture.  You have to run this on ARM directly, like a Raspberry Pi.

So Focas ARM library will run on armv7 and aarch64.  Both of which have supported .NET.
Focas Linux x86 will run on 32 bit Linux, but .NET is not supported on Linux 32 bit.
Focas Linux x64 will run on 64 bit Linux, but we have the issues with unmanaged code marshaling.
... and Focas Windows 64bit will run on 64bit Windows, but again, unmanaged code marshaling issues.

And you don't need `docker push`.  That pushes the built image to https://hub.docker.com/repository/docker/ladder99/fanuc-driver.

Also, you should not `sudo` `docker run`.

tukusejssirs, 2021-06-10:
> I see. Would it be possible to use `fwlib64.cs` without (much) modifications in `fanuc-driver`? I could test it it would work like that, but could you create a simple test script (or a test branch) [...]

@MRIIOT, you haven’t reacted on this yet. Is it possible?

MRIIOT, 2021-06-10:
> @MRIIOT, you haven’t reacted on this yet. Is it possible?

We would have to test each 64bit call individually.  So yes, it will be possible.

MRIIOT, 2021-06-10:
> > Aah, you're trying to run an ARMv7 image on x86 architecture. You have to run this on ARM directly, like a Raspberry Pi.
> > So Focas ARM library will run on armv7 and aarch64. Both of which have supported .NET.
> > Focas Linux x86 will run on 32 bit Linux, but .NET is not supported on Linux 32 bit.
> > Focas Linux x64 will run on 64 bit Linux, but we have the issues with unmanaged code marshaling.
> > ... and Focas Windows 64bit will run on 64bit Windows, but again, unmanaged code marshaling issues.
> 
> I’m a bit lost here. Does this mean it won’t work on x86_64 Linux?

Only the Linux 64 Fanuc library will work in the x86_64 context, bare metal or docker.

Only the Arm Focas library will work in the ARMV7/AARCH context.

> 
> > And you don't need `docker push`. That pushes the built image to https://hub.docker.com/repository/docker/ladder99/fanuc-driver.
> 
> So remove it from the instructions please. 😉 You place a hash sign before it, but I though it means to be run as superuser, but now I think use just commented it out.
> 

OK

> > Also, you should not `sudo` `docker run`.
> 
> Well, yeah, but the default installation of `docker` via `dnf`/`yum` on RHEL-like systems allows running it only by root. Of course, I could create a separate user/group (like you suggest in the instructions), but for testing purposes this shouldn’t an issue.

tukusejssirs, 2021-06-10:
> > > I see. Would it be possible to use `fwlib64.cs` without (much) modifications in `fanuc-driver`? I could test it it would work like that, but could you create a simple test script (or a test branch) for me that would utilise `fwlib64.cs`? What I mean is when running on x86_64 OS, simply use `fwlib64.cs`.
> > 
> > 
> > @MRIIOT, you haven’t reacted on this yet. Is it possible?
> 
> We would have to test each 64bit call individually. So yes, it will be possible once we validate that 64bit calls are working. It might be more work than necessary right now. I would prefer to build some use cases for 32bit for now... since it is working.

I see. How could I help you validate if the 64-bit calls are working?

MRIIOT, 2021-06-11:
Well.... It's a process.

Here is the result on Ubuntu 20.04.

[image]

[image]

So there is no need for a separate fwlib64.cs.  The fwlib32.cs needs to be modified to work with 32 and 64 bit.  Above is an example of a call to cnc_rdparam with IODBPSD_1 structure.  

The changes required just for this one are:

before:

```c#
[StructLayout(LayoutKind.Explicit)]
        public class IODBPSD_1
        {
            [FieldOffset(0)]
            public short datano;
            [FieldOffset(2)]
            public short type;
            [FieldOffset(4)]
            public byte cdata;
            [FieldOffset(4)]
            public short idata;
            [FieldOffset(4)]
            public int ldata;
        }
```
after:
```c#
        [StructLayout(LayoutKind.Sequential)]
        public class IODBPSD_1
        {
            public short datano;
            public short type;
            public IODBPSD_1_INNER data;
        }

        [StructLayout(LayoutKind.Explicit)]
        public class IODBPSD_1_INNER
        {
            [FieldOffset(0)]
            public byte cdata;
            [FieldOffset(0)]
            public short idata;
            [FieldOffset(0)]
            public int ldata;
        }
```

This of course trickles down because now any references to IODBPSD_1.ldata are now IODBPSD_1.data.ldata.

I guess validation of data is best done at the machine depending on the data we want to collect.  So we can focus on any of the current Focas calls I have implemented.

Someone (I can when I have time) has to go through the entire fwlib32.cs and:

1.  look for any functions currently being used in the code (marked with LINUX64)
2. find structures that are used in the function call
[image]
3. fix up the structures to work with 64bit (like above), test, repeat.

tukusejssirs, 2021-06-11:
Thanks. :+1:

If I got it right, there 240 different functions/classes (see the list below) that need to be fixed + 4 which are commented out (L8788-8799: `IODBPSD_A`, `IODBPSD_B`, `IODBPSD_C`, `IODBPSD_D`). It is quite a lot to do; I presume it cover most (if not all) of the code in `fwlib32.cs`.

<details>
   <summary>list of functions/classes</summary>

```bash
ALMINFO_1
ALMINFO_2
CFILEINFO
DSMNTINFO
HSPDATA_1
HSPDATA_2
HSPDATA_3
HSPINFO
HSPINFO_data
IDBCHAN
IDBITD
IDBMGRP
IDBPDFSDIR
IDBPDFTDIR
IDBSFBCHAN
IDBTLM
IDBUNSOLICMSG
IDBWRA
IDBWRC
IDBWRR
IODBBUSPRM
IODBDIDO
IODBDSSET
IODBETP_ETB
IODBETP_FTP
IODBETP_HOST
IODBETP_TCP
IODBGNRL
IODBHPAC
IODBHPPR
IODBHPST
IODBIDINF
IODBINDEADR
IODBITEM
IODBMLTTL
IODBMR
IODBMSTP
IODBMTAP
IODBOVL
IODBPI
IODBPMAINTE
IODBPMC0
IODBPMC1
IODBPMC2
IODBPMCCNTL
IODBPMCEXT
IODBPR
IODBPRFADR
IODBPRM
IODBPRMNO
IODBPSD_1
IODBPSD_2
IODBPSD_3
IODBPSD_4
IODBRDNA
IODBRMTPRM
IODBSAFE
IODBSGNL
IODBSIG
IODBSIG2
IODBSIG3
IODBSIGAD
IODBSLVADR
IODBSLVID
IODBSLVPRM
IODBSLVPRM2
IODBSLVPRM3
IODBTD
IODBTD2
IODBTGI
IODBTGI2
IODBTGI3
IODBTGI4
IODBTIME
IODBTLCTL
IODBTLDT
IODBTLGRP
IODBTLMAG
IODBTLMAG2
IODBTLMNG
IODBTLMNG_F2
IODBTLMNG_F2_data
IODBTLTOOL
IODBTLZN
IODBTO_1_1
IODBTO_1_2
IODBTO_1_3
IODBTO_2
IODBTO_3
IODBTR
IODBUNSOLIC
IODBWAVE
IODBWVPRM
IODBZOFS
IODBZOR
ODB3DCD
ODB3DHDL
ODB3DPLS
ODB3DTO
ODB5AXMAN
ODBACT
ODBACT2
ODBACTTLZN
ODBAD
ODBAHIS
ODBAHIS2
ODBAHIS3
ODBAHIS5
ODBALM
ODBALMMSG
ODBALMMSG2
ODBAXDT
ODBAXIS
ODBAXISNAME
ODBBRS
ODBBUF
ODBCAXIS
ODBCMD
ODBCSS
ODBDATRNG
ODBDGN_1
ODBDGN_2
ODBDGN_3
ODBDGN_4
ODBDIAGIF
ODBDIAGNUM
ODBDNCDGN
ODBDSDIR
ODBDSMNT
ODBDY2_1
ODBDY2_2
ODBERR
ODBETMSG
ODBEXAXISNAME
ODBEXEM
ODBEXEPRG
ODBEXGP
ODBFINFO
ODBFINFORM
ODBFIX
ODBGCD
ODBHDDDIR
ODBHDDINF
ODBHIS
ODBHND
ODBHOSTDIR
ODBLFNO
ODBLOFS
ODBM
ODBMDIP
ODBMDL_1
ODBMDL_2
ODBMDL_3
ODBMDL_4
ODBMDL_5
ODBMGRP
ODBMIR
ODBMVINF
ODBNC_1
ODBNC_2
ODBNODE
ODBOPHIS4_1
ODBOPHIS4_10
ODBOPHIS4_11
ODBOPHIS4_2
ODBOPHIS4_3
ODBOPHIS4_4
ODBOPHIS4_5
ODBOPHIS4_6
ODBOPHIS4_7
ODBOPHIS4_8
ODBOPHIS4_9
ODBPARAIF
ODBPARANUM
ODBPDFNFIL
ODBPDFSDIR
ODBPM
ODBPMCADR
ODBPMCALM
ODBPMCERR
ODBPMCINF
ODBPMCTITLE
ODBPMINF
ODBPOFS
ODBPOS
ODBPRFCNF
ODBPRO
ODBPRS
ODBPSER
ODBPTIME
ODBPTLINF
ODBRMTDT
ODBROFS
ODBROT
ODBRSTRM
ODBSCL
ODBSD
ODBSEQ
ODBSETIF
ODBSETNUM
ODBSINFO
ODBSLVST
ODBSPDI
ODBSPDLNAME
ODBSPDO
ODBSPEED
ODBSPLOAD
ODBSPN
ODBSRCSLYT
ODBSRCSST
ODBST
ODBSVLOAD
ODBSYS
ODBSYSC
ODBSYSEX
ODBSYSH
ODBSYSS3
ODBTG
ODBTLIFE1
ODBTLIFE2
ODBTLIFE3
ODBTLIFE4
ODBTLIFE5
ODBTLINF
ODBTLINF2
ODBTLINFO
ODBTLUSE
ODBTOFS
ODBUP
ODBUP3
ODBUSEGR
ODBUSEGRP
ODBWVDT
OPMSG
OPMSG2
OPMSG3
PRGDIR
PRGDIRTM
PRGPNT
SRAMADDR
```

</details>

> I see this pattern across a lot function calls in the Focas library. Here, for example, parameters can be different sizes, so they just stick cdata, idata, ldata into the same structure, same memory location, just different size.

Okay, therefore I need to get the parameter list in order fix those functions/methods, right?

> It's up to the developer to know to look at ldata in the case of 'power on time'.

Therefore when I want to implement `power on time` in my app (which would use `fanuc-driver`), I need to check the parameter list of the particular controller to get the right variable name (`cdata`, `idata`, or `ldata`), right?

MRIIOT, 2021-06-11:
> Thanks. 👍
> 
> If I got it right, there 240 different functions/classes (see the list below) that need to be fixed + 4 which are commented out (L8788-8799: `IODBPSD_A`, `IODBPSD_B`, `IODBPSD_C`, `IODBPSD_D`). It is quite a lot to do; I presume it cover most (if not all) of the code in `fwlib32.cs`.
> 
> list of functions/classes

Yeah, and I don't think we have to touch them all right away, just as they are needed.

> Okay, therefore I need to get the parameter list in order fix those functions/methods, right?

Yes, should be in the manual, called out with data type.  I think there is a Focas function to interrogate a specific parameter and return what type it is and stuff.

> Therefore when I want to implement `power on time` in my app (which would use `fanuc-driver`), I need to check the parameter list of the particular controller to get the right variable name [...]

Yes sir.

MRIIOT, 2021-06-12:
> 
> If I got it right, there 240 different functions/classes (see the list below) that need to be fixed + 4 which are commented out (L8788-8799: `IODBPSD_A`, `IODBPSD_B`, `IODBPSD_C`, `IODBPSD_D`). It is quite a lot to do; I presume it cover most (if not all) of the code in `fwlib32.cs`.

I think we can use this list to keep track of what has been converted to 64bit and tested .

tukusejssirs, 2021-06-14:
Argh … I’m a bit lost in C#; so many new stuff for me.

---

Is the following code refactoring the right way to do it? Note that I didn’t check anything anywhere nor changed anything else in the code. I seems I need to changed something else, because the `collector sweep failed` didn’t go away.

```c#
// Original code
[StructLayout(LayoutKind.Explicit)]
public class IODBPSD_3
{
   [FieldOffset(0)]
   public short datano;    /* data number */
   [FieldOffset(2)]
   public short type;      /* axis number */
   [FieldOffset(4),
   MarshalAs(UnmanagedType.ByValArray, SizeConst = MAX_AXIS)]
   public byte[] cdatas = new byte[MAX_AXIS];
   [FieldOffset(4),
   MarshalAs(UnmanagedType.ByValArray, SizeConst = MAX_AXIS)]
   public short[] idatas = new short[MAX_AXIS];
   [FieldOffset(4),
   MarshalAs(UnmanagedType.ByValArray, SizeConst = MAX_AXIS)]
   public int[] ldatas = new int[MAX_AXIS];
}

// Refactored code
[StructLayout(LayoutKind.Sequential)]
public class IODBPSD_3
{
   public short datano;
   public short type;
   public IODBPSD_3_INNER data;
}

[StructLayout(LayoutKind.Explicit)]
public class IODBPSD_3_INNER
{
   [FieldOffset(0),
   MarshalAs(UnmanagedType.ByValArray, SizeConst = MAX_AXIS)]
   public byte[] cdatas = new byte[MAX_AXIS];
   [FieldOffset(0),
   MarshalAs(UnmanagedType.ByValArray, SizeConst = MAX_AXIS)]
   public short[] idatas = new short[MAX_AXIS];
   [FieldOffset(0),
   MarshalAs(UnmanagedType.ByValArray, SizeConst = MAX_AXIS)]
   public int[] ldatas = new int[MAX_AXIS];
}
```

> edit: unions .... https://docs.microsoft.com/en-us/dotnet/framework/interop/marshaling-classes-structures-and-unions#unions-sample

Unions look be me like simple objects in JS or hashes in Perl, but unions are typed, right?

Anyway, where do I need them to use? I grepped the whole code, but no match found.

> > Okay, therefore I need to get the parameter list in order fix those functions/methods, right?
> 
> edit: not really, the method signatures are what they are.

What are _method signatures_?

> the cnc_rdparam function that takes IODBPSD_1 is fixed up for 64bit. IODBPSD_2, IODBPSD_3, IODBPSD_4 should be fixed up as they can be used right now from fanuc-driver code base.

At least for me, the latest commit of `main` still complains about `collector sweeping failed` and that it could not load `IODBPSD_3`.

> edit: ObjectLayoutInspector... https://github.com/SergeyTeplyakov/ObjectLayoutInspector

How can I use `ObjectLayoutInspector` with `fanuc-driver` (actually with the FOCAS2 response)?

> Yes, should be in the manual, called out with data type. I think there is a Focas function to interrogate a specific parameter and return what type it is and stuff.

That function might be useful for me (us) to get the parameter metadata, but I have no idea how to use it.

tukusejssirs, 2021-06-14:
Meanwhile, I have created a list of data we would like to get (read) from Fanucs with functions from the list provided by [inventcom.net](https://www.inventcom.net/fanuc-focas-library/general/flist_all). I did my best to assign those functions correctly; you might find some misassignments and definitely there are some missing functions which provide some specific information.

- what we are actually interested in to get (read):
   - machine information/metadata:
      - vendor, model, controller model, serial number, type (lathe/mill/etc), ethernet (TCP/IP) configuration, hardware information, any additional metadata:
         - `cnc_sysinfo`;
         - `cnc_sysinfo_ex` (difference between `cnc_sysinfo` and `cnc_sysinfo_ex`?);
         - `cnc_sysconfig`;
         - `cnc_rdsyssoft`;
         - `cnc_rdsyssoft2`;
         - `cnc_rdsyssoft3`;
         - `cnc_rdsyshard`;
         - `cnc_rdetherinfo`;
   - production:
      - currently running program and subprogram (if any) name + additional metadata (if any);
         - `cnc_rdproginfo`;
         - `cnc_rdprgnum`;
         - `cnc_exeprgname`;
         - `cnc_rdseqnum`;
         - `cnc_rdblkcount`;
         - `cnc_rdexecprog`;
         - `cnc_rdrepeatval`;
         - `cnc_rdrepeatval_ext`;
      - when a program started and when ended;
      - when a particular part was started to be produced and when ended (this is just nice to have);
      - parts count (total per program to be produced and count of parts already produced);
         - using `cnc_rdparam` (`6711`, `6712` and `6713`);
   - machine status:
      - alarm codes and messages (ideally with timestamp when they started and ended + any available metadata like where the alarm is located or alarm severity):
         - `cnc_alarm`;
         - `cnc_alarm2`;
         - `cnc_rdalminfo`;
         - `cnc_rdalmmsg` (this one may be redundant, as `cnc_rdalmmsg2` should contain the same information + axis names);
         - `cnc_rdalmmsg2`;
         - `cnc_getdtailerr` (I am not quite sure if the _detailed error_ is regarding alarms or some other kind of errors);
      - what state is the machine (automatic/manual/whatever + any available metadata):
         - `cnc_statinfo`;
         - `cnc_statinfo2`;
      - feedrate override;
         - `cnc_rdopnlsgnl` (`slct_data`, bits 4 and 5);
      - rapid override (this is probably called `rapid traverse override` by Fanuc; this is not that important to get);
         - `cnc_rdopnlsgnl` (`slct_data`, bit 3).

tukusejssirs, 2021-06-15:
Okay, with the latest code (`main` @ 337aded) and the changing of the code below in [`fwlib.cs` (L1240-1256)](https://github.com/Ladder99/fanuc-driver/blob/main/fanuc/fwlib.cs#L1240:L1256) there is no error, but also no output (beside the values of `--config` and `--nlog`).

```c#
// Original code
[StructLayout(LayoutKind.Explicit)]
public class IODBPSD_3
{
   [FieldOffset(0)]
   public short datano;    /* data number */
   [FieldOffset(2)]
   public short type;      /* axis number */
   [FieldOffset(4),
   MarshalAs(UnmanagedType.ByValArray, SizeConst = MAX_AXIS)]
   public byte[] cdatas = new byte[MAX_AXIS];
   [FieldOffset(4),
   MarshalAs(UnmanagedType.ByValArray, SizeConst = MAX_AXIS)]
   public short[] idatas = new short[MAX_AXIS];
   [FieldOffset(4),
   MarshalAs(UnmanagedType.ByValArray, SizeConst = MAX_AXIS)]
   public int[] ldatas = new int[MAX_AXIS];
}

// Refactored code
[StructLayout(LayoutKind.Sequential)]
public class IODBPSD_3
{
   public short datano;
   public short type;
   public IODBPSD_3_INNER data;
}

[StructLayout(LayoutKind.Explicit)]
public class IODBPSD_3_INNER
{
   [FieldOffset(0),
   MarshalAs(UnmanagedType.ByValArray, SizeConst = MAX_AXIS)]
   public byte[] cdatas = new byte[MAX_AXIS];
   [FieldOffset(0),
   MarshalAs(UnmanagedType.ByValArray, SizeConst = MAX_AXIS)]
   public short[] idatas = new short[MAX_AXIS];
   [FieldOffset(0),
   MarshalAs(UnmanagedType.ByValArray, SizeConst = MAX_AXIS)]
   public int[] ldatas = new int[MAX_AXIS];
}
```

How can I output the gathered values?

tukusejssirs, 2021-06-15:
Anyway, could the `focas2.log` be a text file with newline-delimited records? Currently, it contains some control characters (at least `^C`/`ETX`/`0x03`) and the records are delimited with 175+ spaces. From my point of view, it is a log and as such it should be a bit more human readable.

For now, I use `sed 's/\x03//g;s/\r/\n/g;s/ \+\([0-9]\{4\}\/[0-9]\{2\}\/[0-9]\{2\}\)/\n\1/g' focas2.log | sed -z 's/^\n*//;s/\n*$//' | less` to make it more human-readable.

MRIIOT, 2021-06-15:
I'm traveling this week.  I will be back later this week.  I will probably hop on tomorrow night.

As far as output, turn up the logging in nlog.config.  Also, do you have an mqtt broker set up?  That is where the data goes by default.  What system will be consuming this data in your environment?

MRIIOT, 2021-06-16:
> Anyway, could the `focas2.log` be a text file with newline-delimited records? Currently, it contains some control characters (at least `^C`/`ETX`/`0x03`) and the records are delimited with 175+ [...]

focas2.log is a Linux only log from the Fanuc library.  We have no control over it.

`public static extern short cnc_startupprocess(long level, string filename);`

MRIIOT, 2021-06-16:
> Okay, with the latest code (`main` @ [337aded](https://github.com/Ladder99/fanuc-driver/commit/337aded08552d931905a26775693033577c8dc9a)) and the changing of the code below in [`fwlib.cs` (L1240-1256)](https://github.com/Ladder99/fanuc-driver/blob/main/fanuc/fwlib.cs#L1240:L1256) there is no error, but also no output (beside the values of `--config` and `--nlog`).
> 
> ```cs
> 
> public class IODBPSD_3_INNER
> {

I renamed _INNER to _UNION, which makes more sense.

tukusejssirs, 2021-06-16:
> do you have an mqtt broker set up? That is where the data goes by default.

No, I didn’t. I coudn’t find anything on MQTT configuration in `fanuc-driver` in `README.md`. Could you provide me a sample configuration to subscribe to all messages/topics?

Anyway, could you tell me what is that `disco` for in MQTT? I could not find anything on this on the Internet (it looks like I queried the wrong strings :wink:).

> What system will be consuming this data in your environment?

Well, my downstream app would be written in Node JS (JS for now, later in TS). It would process the gathered data. All of this (`fanuc-driver` + my Node JS app) would run locally on a x86_64 Linux server.

> focas2.log is a Linux only log from the Fanuc library. We have no control over it.
> 
> `public static extern short cnc_startupprocess(long level, string filename);`

I see. But could you not catch the output prior to writing it to file? I know, this is not _that_ important, but it’s nice to have. :wink:

> I renamed `_INNER` to `_UNION`, which makes more sense.

Thanks for letting me know. :+1:

So all those unions are actually _objects_ (as in JSON/TS/TS) or _hashes_ (as in Perl), right?

---

Could you make `fanuc-driver` work from any CWD? You should use the script/program path as the CWD of the program, not the CWD the user is in when they run `fanuc-driver`. See the log below.

<details>
   <summary><code>dotnet run</code></summary>

```bash
CSC : error CS2001: Source file '/home/ts/git/c10n/fanuc_driver_ts/fanuc/../base-driver/base/Collector.cs' could not be found. [/home/ts/git/c10n/fanuc_driver_ts/fanuc/fanuc.csproj]
CSC : error CS2001: Source file '/home/ts/git/c10n/fanuc_driver_ts/fanuc/../base-driver/base/Handler.cs' could not be found. [/home/ts/git/c10n/fanuc_driver_ts/fanuc/fanuc.csproj]
CSC : error CS2001: Source file '/home/ts/git/c10n/fanuc_driver_ts/fanuc/../base-driver/base/Machine.cs' could not be found. [/home/ts/git/c10n/fanuc_driver_ts/fanuc/fanuc.csproj]
CSC : error CS2001: Source file '/home/ts/git/c10n/fanuc_driver_ts/fanuc/../base-driver/base/Machines.cs' could not be found. [/home/ts/git/c10n/fanuc_driver_ts/fanuc/fanuc.csproj]
CSC : error CS2001: Source file '/home/ts/git/c10n/fanuc_driver_ts/fanuc/../base-driver/base/mqtt/Broker.cs' could not be found. [/home/ts/git/c10n/fanuc_driver_ts/fanuc/fanuc.csproj]
CSC : error CS2001: Source file '/home/ts/git/c10n/fanuc_driver_ts/fanuc/../base-driver/base/mqtt/Brokers.cs' could not be found. [/home/ts/git/c10n/fanuc_driver_ts/fanuc/fanuc.csproj]
CSC : error CS2001: Source file '/home/ts/git/c10n/fanuc_driver_ts/fanuc/../base-driver/base/mqtt/Disco.cs' could not be found. [/home/ts/git/c10n/fanuc_driver_ts/fanuc/fanuc.csproj]
CSC : error CS2001: Source file '/home/ts/git/c10n/fanuc_driver_ts/fanuc/../base-driver/base/mqtt/SparkplugB.cs' could not be found. [/home/ts/git/c10n/fanuc_driver_ts/fanuc/fanuc.csproj]
CSC : error CS2001: Source file '/home/ts/git/c10n/fanuc_driver_ts/fanuc/../base-driver/base/Veneer.cs' could not be found. [/home/ts/git/c10n/fanuc_driver_ts/fanuc/fanuc.csproj]
CSC : error CS2001: Source file '/home/ts/git/c10n/fanuc_driver_ts/fanuc/../base-driver/base/Veneers.cs' could not be found. [/home/ts/git/c10n/fanuc_driver_ts/fanuc/fanuc.csproj]
```

---

I have updated OP with the list of functions/classes.

MRIIOT, 2021-06-16:
> > do you have an mqtt broker set up? That is where the data goes by default.
> 
> 
> 
> No, I didn’t. I coudn’t find anything on MQTT configuration in `fanuc-driver` in `README.md`. Could you provide me a sample configuration to subscribe to all messages/topics?
> 

Look in config.yml at the broker section.

tukusejssirs, 2021-06-16:
> Look in config.yml at the broker section.

Yes, I have looked in there, see the config below.

```yaml
broker:
  enabled: !!bool true
  net_ip: 127.0.0.1
  net_port: !!int 1883
  auto_connect: !!bool true
  publish_status: !!bool true
  publish_arrivals: !!bool true
  publish_changes: !!bool true
  publish_disco: !!bool true
  disco_base_topic: fanuc
```

I have started an MQTT server (broker) in Node JS using [`aedes`](https://github.com/moscajs/aedes/blob/main/docs/Examples.md); I believe I need to subscribe some topics/messages, but I have no idea what are their name (or whatever they are called). Or should I start an MQTT client using [`async-mqtt`](https://github.com/mqttjs/async-mqtt)? I am a bit lost in the MQTT terminology (I have never used MQTT).

tukusejssirs, 2021-06-17:
`fanuc-driver` seems to connect to the MQTT broker (server), because when the broker is not running prior to starting `fanuc-driver`, it errors out that the `broker connection failed`. When the broker is already running, there is no error, but also no output.

This underlines my assumptions that all I need is to subscribe some topics/messages. Could you tell me where could I find the topic/message names I need to subscribe? Thanks in advance! :wink:

MRIIOT, 2021-06-19:
> 
> A bit OT (related to #8): as I understand it, `fanuc-driver` gathers all information it can (i.e. it is programmed to collect). Could we define what should be gathered in order to minimise the communication between machine(s) and `fanuc-driver`? Of course, there are some data that should be always gathered (esp at connection initialisation), but even most of that could be configurable in `config.yml` (i.e. the downstream app could collect those information `fanuc-driver` runs the very first time and the subsequent `fanuc-driver` runs could use that info instead of collecting them again. Of course, this is just a brain exercise for you.

Agreed, no need to pull axis names every second.  This should be optimized eventually.

MRIIOT, 2021-06-19:
> > What system will be consuming this data in your environment?
> 
> Well, my downstream app would be written in Node JS (JS for now, later in TS). It would process the gathered data. All of this (`fanuc-driver` + my Node JS app) would run locally on a x86_64 Linux server.

MQTT-WS is your friend.

> > focas2.log is a Linux only log from the Fanuc library. We have no control over it.
> > `public static extern short cnc_startupprocess(long level, string filename);`
> 
> I see. But could you not catch the output prior to writing it to file? I know, this is not _that_ important, but it’s nice to have. 😉
> 

It looks like it would be helpful for reversing the protocol on the wire.

> Could you make `fanuc-driver` work from any CWD? You should use the script/program path as the CWD of the program, not the CWD the user is in when they run `fanuc-driver`. See the log below.
> 
> `dotnet run`

How?

MRIIOT, 2021-06-19:
> `fanuc-driver` seems to connect to the MQTT broker (server), because when the broker is not running prior to starting `fanuc-driver`, it errors out that the `broker connection failed`. When the broker is already running, there is no error, but also no output.

Yup, re-connects need to be handled.

tukusejssirs, 2021-06-19:
> MQTT-WS is your friend.

You mean [`mqtt-ws`](https://www.npmjs.com/package/mqtt-ws)? That code seen last release 8 years ago. The repo link is dead. The NPM description says is uses [`mqtt.js`](https://github.com/mqttjs/MQTT.js) which seems to be client only, while (as I understand it), `fanuc-driver` is implemented as MQTT client and the client app needs to be an MQTT server, which is why I use [`aedes`](https://github.com/moscajs/aedes), which works as expected.

Therefore currently, I don’t see this as an issue. :wink:

> > > focas2.log is a Linux only log from the Fanuc library. We have no control over it.
> > > `public static extern short cnc_startupprocess(long level, string filename);`
> > 
> > I see. But could you not catch the output prior to writing it to file? I know, this is not _that_ important, but it’s nice to have. wink
> 
> It looks like it would be helpful for reversing the protocol on the wire.

Indeed it would be useful for that. :wink:

> > Could you make `fanuc-driver` work from any CWD? You should use the script/program path as the CWD of the program, not the CWD the user is in when they run `fanuc-driver`. See the log below.
> > `dotnet run`
> 
> How?

You could take a look at [this SO answer](https://stackoverflow.com/a/11882118/3408342) or other answers in that question.

Basically, all you need to do is:

1. Get the path to the folder with the project file (`{repo_root}/fanuc`).
2. Prepend that path to any and all paths used in the code.

That way you wouldn’t use relative paths anymore, but absolute paths.

tukusejssirs, 2021-06-23:
As I have no idea how to test individual functions/classes, I tried to run `Basic02` strategy type. It works if I comment out [L97-98](https://github.com/Ladder99/fanuc-driver/blob/main/fanuc/collectors/Basic02.cs#L97:L98), however I wasn’t able to locate the function/class that is used there.  How can I know from the following output which function/class I need to modify?

Anyway, could prepare for me a collector (or whatever is needed) to get the [data required by me](https://github.com/Ladder99/fanuc-driver/issues/4#issuecomment-860846820) in order to test it on Linux x86_66 and try to make it work there? Or at least some portion of it for now. Thanks in advance!

MRIIOT, 2021-06-23:
> As I have no idea how to test individual functions/classes, I tried to run `Basic02` strategy type. It works if I comment out [...]

https://github.com/Ladder99/fanuc-driver/blob/main/fanuc/collectors/Basic02.cs

Ok, this is a really bad analogy but here it goes.

In a `Collector` class, the [InitializeAsync](https://github.com/Ladder99/fanuc-driver/blob/0d59fb471a3d197b1b7c021229b3129aa947e724/fanuc/collectors/Basic04.cs#L42) method is used to define what stickers we want to put our machine using the various `Apply...` methods.  It can be a sticker of a cat, because we like cats.  When we look at the sticker we are looking at (observing) a cat.  A cat is made of up of four legs, tail, head, whatever.  The `Collector` the glue that holds our sticker to the machine.

Here is a basic [cat sticker](https://github.com/Ladder99/fanuc-driver/blob/main/fanuc/veneers/StatInfo.cs).
Here is the same basic [cat sticker but different](https://github.com/Ladder99/fanuc-driver/blob/main/fanuc/veneers/StatInfoText.cs).

```lua
function script:init_paths(this, collector)
    print("initialize paths");
    
    collector:apply("StatInfo","stat_1");
    collector:apply("StatInfoText","stat_2");
end
```

What lets us see the cat on the sticker are all the parts that make up the cat.  We need to make some Focas calls first.  In this case [`cnc_statinfo` Focas function call](https://github.com/Ladder99/fanuc-driver/blob/main/fanuc/platform/StatInfo.cs) is pretty much a 1:1 representation of a cat.

```lua
function script:collect_path(this, collector, current_path)
    print("collect path " .. current_path);
    
    stat1 = collector.Platform:StatInfo();
    print(collector:to_json(stat1));
end
```

The data from the Focas call doesn't look like much of a cat, it's more like the cat's DNA.

```json
{
  "method": "cnc_statinfo",
  "invocationMs": 4,
  "doc": "https://www.inventcom.net/fanuc-focas-library/misc/cnc_statinfo",
  "success": true,
  "rc": 0,
  "request": {
    "cnc_statinfo": {}
  },
  "response": {
    "cnc_statinfo": {
      "statinfo": {
        "dummy": 0,
        "tmmode": 0,
        "aut": 3,
        "run": 0,
        "motion": 0,
        "mstb": 0,
        "emergency": 0,
        "alarm": 0,
        "edit": 0
      }
    }
  }
}
```

To make it look like a cat, [we peel the sticker](https://github.com/Ladder99/fanuc-driver/blob/0d59fb471a3d197b1b7c021229b3129aa947e724/fanuc/collectors/Basic04.cs#L132) using the various `Peel...` methods and take a closer look.  The [`Handler` class](https://github.com/Ladder99/fanuc-driver/blob/0d59fb471a3d197b1b7c021229b3129aa947e724/fanuc/handlers/Native.cs#L52) is our hand bringing the sticker close to our face so we can see it.

```lua
function script:collect_path(this, collector, current_path)
    print("collect path " .. current_path);
    
    stat1 = collector.Platform:StatInfo();
    print(collector:to_json(stat1));
    collector:set_native_and_peel("stat_1", stat1);
    
    stat2 = collector.Platform:StatInfo();
    print(collector:to_json(stat2));
    collector:set_native_and_peel("stat_2", stat2);
end
```

One cat sticker looks like this:

```json
{
  "observation": {
    "time": 1624455587088,
    "machine": "sim_lua",
    "type": "source",
    "name": "stat_1",
    "marker": {
      "path_no": 1
    }
  },
  "source": {
    "method": "cnc_statinfo",
    "invocationMs": 6,
    "data": {}
  },
  "state": {
    "time": "00:00:02.5161228",
    "data": {
      "mode": {
        "automatic": 3
      },
      "status": {
        "run": 0,
        "edit": 0,
        "motion": 0,
        "mstb": 0,
        "emergency": 0,
        "alarm": 0
      }
    }
  }
}
```

The other cat sticker looks like this:

```json
{
  "observation": {
    "time": 1624455587136,
    "machine": "sim_lua",
    "type": "source",
    "name": "stat_2",
    "marker": {
      "path_no": 1
    }
  },
  "source": {
    "method": "cnc_statinfo",
    "invocationMs": 6,
    "data": {}
  },
  "state": {
    "time": "00:00:02.5583945",
    "data": {
      "mode": {
        "automatic": "MEMory"
      },
      "status": {
        "run": "STOP",
        "edit": "****(Not editing)",
        "motion": "***",
        "mstb": "***",
        "emergency": "(Not emergency)",
        "alarm": "(No alarm)"
      }
    }
  }
}
```

Stickers can have more stuff on them, like a zoo.  The [SpindleData](https://github.com/Ladder99/fanuc-driver/blob/main/fanuc/veneers/SpindleData.cs) sticker combines a bunch of Focas data into one sticker.  [Here](https://github.com/Ladder99/fanuc-driver/blob/main/fanuc/collectors/SpindleData.cs) is the glue.

```
await Peel("spindle_data", 
                current_spindle,
                Get("spindle_names"), 
                Get("sp_speed"), 
                Get("sp_meter"), 
                Get("sp_maxrpm"), 
                Get("sp_gear"),
                Get("diag_lnk"),
                Get("diag_temp"),
                Get("diag_comms"),
                Get("diag_load_perc"),
                Get("diag_load_min"),
                Get("diag_coder"),
                Get("diag_loop_dev"),
                Get("diag_sync_error"),
                Get("diag_pos_data"),
                Get("diag_error"),
                Get("diag_warn"),
                Get("diag_rev_1"),
                Get("diag_rev_2"));
```

[image]

> Anyway, could prepare for me a collector (or whatever is needed) to get the [data required by me](https://github.com/Ladder99/fanuc-driver/issues/4#issuecomment-860846820) in order to test it on Linux x86_66 and try to make it work there? Or at least some portion of it for now. Thanks in advance!

Your use case makes sense and it will be put together.  I'll rely on you to test on Linux x86_64 and we can patch what needs to be patched as we go along.

tukusejssirs, 2021-06-23:
Thanks for an awesome, in-depth description of how `fanuc-driver` works and the craziest cat analogy I’ve ever heard! :rofl: :+1: I still need to make some test it out myself to get familiar with it.

Anyway, as far as I get it right from your comment, a `collector` is rather a `collection`, isn’t it?

---

OT: Is the casing of the strings in [`StatInfoText.cs` (L22, L23, L27-28)](https://github.com/Ladder99/fanuc-driver/blob/main/fanuc/veneers/StatInfoText.cs#L22:L28) (and elsewhere in that file) intentional? Should it not rather be changed (e.g.) from `STaRT` to `START`? It might be though you wanted to say that `start` is shortened to `STRT`, but this way it looks kind of strange. :wink:

OT: There can be only a single alarm triggered in any single moment? In [`StatInfoText.cs` (L91)](https://github.com/Ladder99/fanuc-driver/blob/main/fanuc/veneers/StatInfoText.cs#L91), it is defined as a string, however, I’d expect an array of strings.

MRIIOT, 2021-06-23:
> Thanks for an awesome, in-depth description of how `fanuc-driver` works and the craziest cat analogy I’ve ever heard! 🤣 👍 I still need to make some test it out myself to get familiar with it.
> 
> Anyway, as far as I get it right from your comment, a `collector` is rather a `collection`, isn’t it?
> 

Hmmm, well the Collector does the act of collecting.  It sits between Veneer classes and the machine.

> OT: Is the casing of the strings in [`StatInfoText.cs` (L22, L23, L27-28)](https://github.com/Ladder99/fanuc-driver/blob/main/fanuc/veneers/StatInfoText.cs#L22:L28) (and elsewhere in that file) [...]

That is straight out of the Focas docs. Didn't want to change it.

> OT: There can be only a single alarm triggered in any single moment? In [`StatInfoText.cs` (L91)](https://github.com/Ladder99/fanuc-driver/blob/main/fanuc/veneers/StatInfoText.cs#L91), it is defined as a string, however, I’d expect an array of strings.

No, that bit only signifies that alarm(s) are present.  These two classes read the actual alarms.

https://github.com/Ladder99/fanuc-driver/blob/main/fanuc/platform/RdAlmMsg.cs
https://github.com/Ladder99/fanuc-driver/blob/main/fanuc/platform/RdAlmMsg2.cs

tukusejssirs, 2021-07-12:
> Your use case makes sense and it will be put together. I'll rely on you to test on Linux x86_64 and we can patch what needs to be patched as we go along.

Lately, I’ve been caught up some other work … I’ve just noticed you have updated all the collectors. Have you had some time to puth together a collector for me? You might have been caught up by sometimg else … If you did work on it, could you provide me some details (which collector to use, etc)? Thanks in advance!

MRIIOT, 2021-07-19:
> Lately, I’ve been caught up some other work … I’ve just noticed you have updated all the collectors. Have you had some time to puth together a collector for me? You might have been caught up by [...]

I have not had a chance.  I hope to make progress this coming week.

MRIIOT, 2021-07-26:
@tukusejssirs Give this one a try: https://github.com/Ladder99/fanuc-driver/blob/main/fanuc/collectors/UseCase01.cs

It's not complete, but it's a good start.  Please report your findings.  Recordings using https://pypi.org/project/mqtt-recorder/ and any logs from `fanuc-driver` will be helpful debugging specifics.

tukusejssirs, 2021-07-26:
Thanks, @MRIIOT!

On my first trial, I changed `strategy_type` to `l99.driver.fanuc.collectors.UseCase01, fanuc`, ran MQTT (subscribing to `#`) and I get only `publish` packets (`heartbeat` and `new/clients`) in MQTT (see below) and no error in `fanuc-driver`. 

<details>
   <summary><code>publish</code> packet</summary>

```bash
Packet {
  cmd: 'publish',
  brokerId: 'adfed2d3-dba8-4b49-a844-9abf4451ff66',
  brokerCounter: 3,
  topic: '$SYS/adfed2d3-dba8-4b49-a844-9abf4451ff66/heartbeat',
  payload: 'adfed2d3-dba8-4b49-a844-9abf4451ff66',
  qos: 0,
  retain: false,
  dup: false
}
```

</details>

I also tried `mqtt-recorder`. Although it connected to the broker, no output was saved to file (an empty file was created instead).

MRIIOT, 2021-07-27:
Very cool!

Do you have a multi-path machine?  I'm curious if some data is global or if it is path specific (https://github.com/Ladder99/fanuc-driver/blob/31c6ad63482f939be7261494efc789f163b2f99c/fanuc/collectors/UseCase01.cs#L112).

This packet, controlled by `publish_arrivals: !!bool true` (which enables or disables the *-all topic) shows that no Focas calls failed.  This will become important when we make invalid or unsupported calls.  18 Focas calls take 114ms, and the entire sweep takes over two seconds (two seconds is spent in the C# code itself).  I'm assuming this was the first full sweep since `payload.state.time` is zero. 

```
Packet {
  cmd: 'publish',
  brokerId: 'ac1827a1-c365-4689-a56d-cc327b2e6432',
  brokerCounter: 86,
  topic: 'fanuc/doosan_puma_m9-all/focas_perf',
  payload: '{\n' +
    '  "observation": {\n' +
    '    "time": 1627321494612,\n' +
    '    "machine": "doosan_puma_m9",\n' +
    '    "type": "compound",\n' +
    '    "name": "focas_perf",\n' +
    '    "marker": {}\n' +
    '  },\n' +
    '  "source": {\n' +
    '    "method": "",\n' +
    '    "invocationMs": -1,\n' +
    '    "data": {}\n' +
    '  },\n' +
    '  "state": {\n' +
    '    "time": "00:00:00",\n' +
    '    "data": {\n' +
    '      "sweepMs": 2328,\n' +
    '      "invocation": {\n' +
    '        "count": 18,\n' +
    '        "maxMethod": "cnc_rdparam",\n' +
    '        "maxMs": 23,\n' +
    '        "minMs": 0,\n' +
    '        "avgMs": 6,\n' +
    '        "sumMs": 114,\n' +
    '        "failedMethods": []\n' +
    '      }\n' +
    '    }\n' +
    '  }\n' +
    '}',
  qos: 0,
  retain: true,
  dup: false
}
```

tukusejssirs, 2021-07-27:
> Very cool!
> 
> Do you have a multi-path machine? I'm curious if some data is global

To be honest, I am not a machinist, therefore I have no idea. However, I have found [this](https://exlentedu.com/2020/08/two-path-system-in-fanuc-cnc/) website and I checked the mentioned parameters (`982:S1` → `1`, `982:S2` → `1`, `3717:S1` → `1`, `3717:S2` → `2`), therefore I think it indeed is a two-path/multi-path machine. If I can’t distinguish single and multiple path machines based on these two parameters, tell me how could I do that please. :wink:

> if it is path specific ([`UseCase01.cs` (L112)](https://github.com/Ladder99/fanuc-driver/blob/31c6ad63482f939be7261494efc789f163b2f99c/fanuc/collectors/UseCase01.cs#L112))

I have no idea how I determine this. What do you mean? :wink:

> I'm assuming this was the first full sweep since `payload.state.time` is zero.

Yes, indeed. Otherwise it would not get any data (currently I test `fanuc-driver` with idle machine only).

[10 more replies]

closed 2023-03-10 by MRIIOT