Ladder99/fanuc-driver #7: Use a code formatter

tukusejssirs, 2021-06-15:
I am accustomed to use `prettier` in JS/TS and I see a lot of inconsistency in the code of `fanuc-driver`.

Therefore I suggest to use a code formatter. I have found the following two code formatters for C#:
- [`dotnet-format`](https://github.com/dotnet/format);
- [CSharpier](https://github.com/belav/csharpier).

I would gladly reformat the files in the repo, as long as:
1. you configure the code formatter to your liking;
2. after reformatting, you would abide to the rules.

MRIIOT, 2021-06-15:
Can you give an example?

tukusejssirs, 2021-06-15:
> Can you give an example?

What kind of example?

1. Example configuration?

That depends on which code formatter you would like/agree to use AND what code style (like a guide) you would like to use. I presume you like (or are used to) the ‘Microsoft style’ (see [Coding conventions](https://docs.microsoft.com/en-us/dotnet/csharp/fundamentals/coding-style/coding-conventions) and [Code style rule options](https://docs.microsoft.com/en-us/dotnet/fundamentals/code-analysis/code-style-rule-options)) and therefore you might prefer `dotnet-format`. I have no preference as I don’t code in C# [although I prefer to always put opening brackets (`(`, `[`, `{`) on the previous line, not on a separate line]; all I want is a consistency.

2. Example of what is not consistent?

Somtimes you put blank line between blocks (or before/after a block), sometimes not; some files have a final newline, others not, sometimes you add a space after a comma, elsewhere you omit them, most files use 4 spaces for indentation, sometimes there are only 3 (or one less), …

MRIIOT, 2021-06-15:
Ah, ok yes. Things can be made more consistent.  You didn't say anything about my liberal use of underscores 😀

tukusejssirs, 2021-06-15:
> Ah, ok yes. Things can be made more consistent.

:+1:

> You didn't say anything about my liberal use of underscores :grinning:

No, I didn’t, b/c I don’t code in C#, therefore I don’t exactly know what is a _liberal use_ and what a (sort of) standard. Actually, I don’t care if you create some custom rules if you stick to all the rules, be it custom or standard. :wink: Call me a crazy, I just love consistency and exactness. :neckbeard:

MRIIOT, 2023-01-28:
Default Rider formatting and cleanup rules applied.
fwlib.cs excluded.

closed 2023-01-28 by MRIIOT