Ladder99/fanuc-driver #19: Compare traffic on-the-wire with focas2.log data.

MRIIOT, 2021-07-31:
> > > focas2.log is a Linux only log from the Fanuc library. We have no control over it.
> > > `public static extern short cnc_startupprocess(long level, string filename);`
> > 
> > I see. But could you not catch the output prior to writing it to file? I know, this is not _that_ important, but it’s nice to have. wink
> 
> It looks like it would be helpful for reversing the protocol on the wire.

Indeed it would be useful for that. :wink:

_Originally posted by @tukusejssirs in https://github.com/Ladder99/fanuc-driver/issues/4#issuecomment-864390147_

closed 2022-02-01 by MRIIOT

tukusejssirs, 2022-02-01:
@MRIIOT, I don’t mind marking this issue as `wontfix`, but what is the reason? Especially, what’s the reason for the `security` label?

MRIIOT, 2022-02-01:
Time and priority.  Unless we wanted to reverse the Focas protocol.