Ladder99/fanuc-driver #21: Application goals

MRIIOT, 2021-08-02:
Breaking out into separate conversation.

> > > Something like this? I don't remember if these parameters are global or if each execution path has their own counts (TODO). If each path holds their own counts then it would become `fanuc/{machine}/numberOfPieces/1`, `fanuc/{machine}/numberOfPieces/2`, ...
> > > ```cs
> > > await Peel("numberOfPieces",
> > >     await SetNative("produced", await platform.RdParamDoubleWordNoAxisAsync(6711)),
> > >     await SetNative("producedLife", await platform.RdParamDoubleWordNoAxisAsync(6712)),
> > >     await SetNative("remaining", await platform.RdParamDoubleWordNoAxisAsync(6713)));
> > > ```
> > 
> > 
> > I think yes, something like this.
> 
> What is your db structure and data type to hold this?

The DB structure is not yet decided; it heavily depends on the data we can gather via `fanuc-driver`, but basically each machine will have a separate schema with the following tables:
- `production`:
   - `id`;
   - `start_time`;
   - `end_time`;
   - `program_name`;
   - `parts_count` (how many pieces were produced with a particular program);
- `alarms`:
   - `id`;
   - `start_time`;
   - `end_time`;
   - `code`;
   - `message`;
   - `level` (severity, like info, warning, error);
   - and whatever else we can gather like for example the place where the alarm occurs (e.g. on Mazaks, I know of these values: `MACHINE`, `NC`, `OPERATE`, `PROGRAM`, `SERVO`, `SPINDLE`, `SYSTEM`);
- `status`:
   - light indicator status, i.e. what colours are lit at any given moment;
   - it is gathered every second via a PLC, however, if FOCAS provides a way to get red/yellow/green light status, it would be awesome (I am sceptic though);
   - on Mazaks we also gather feedrate and rapid override level in a separate table (due to the way we gather the data), it might be a good idea to move them into the `status` table.

Anyway, we use PostgreSQL.

> Would you still be breaking out each piece count from the single json?

Well, all in all, we only need the number of pieces produced with a particular program, but sometimes the operators don’t provide how many pieces they need to produce, as they sometimes produce only one piece. In that situation, the pieces to be produced parameter is `0`, but the total number produced [since last Fanuc OS (re-) installation] is incremented. If `fanuc-driver` would take of of this, it would be awesome, but I think the client app should take care of this.

> > As for the question if these params are global, IMHO yes, because one machine can produce only a single piece at a time, but I haven’t checked it anywhere.
> 
> Unless the machine has multiple and independent execution paths, then each one should be able to produce at its own takt.

I see. Then my `production` table should include a path ID, right?

_Originally posted by @tukusejssirs in https://github.com/Ladder99/fanuc-driver/issues/13#issuecomment-891050820_

MRIIOT, 2021-08-02:
> alarms:
id;
start_time;
end_time;

So the end_time would come from the absence of previous alarm/message.

> it is gathered every second via a PLC, however, if FOCAS provides a way to get red/yellow/green light status, it would be awesome (I am sceptic though);

Should be a var we can pull out from Fanuc ladder.  You want to try to find it?

> PostgreSQL

Seems to be good choice.  We are using the same.

> sometimes the operators don’t provide how many pieces they need to produce

Is there work order / labor tracking in another buisness system?

> I see. Then my production table should include a path ID, right?

Yeah, unless you treat it more like a cell and have M9-1, M9-2, and normalize DB further.

tukusejssirs, 2021-08-02:
> So the end_time would come from the absence of previous alarm/message.

Yes, it’d work just like MTConnect. However, `fanuc-driver` does not send a packet yet with empty alarm/message topic when an alarm is cancelled/fixed/dismissed (how do you call it?).

> Should be a var we can pull out from Fanuc ladder. You want to try to find it?

Yes, but I need some clarification: is _Fanuc ladder_ a list in one of the manuals? Or where could I find it? :wink:

Also: do you think it is machine-dependent?

> > sometimes the operators don’t provide how many pieces they need to produce
> 
> Is there work order / labor tracking in another buisness system?

I am not sure what you mean by _another business system_, buy I presume you mean a system in our company used to manage orders and tracking labours. Actually, integration of those two systems (i.e. tracking a part/piece within an order and tracking what part was done by which operator/controller/etc) is the next step. For now we need to gather the production, alarm and status data, then we will move to the next step. :wink:

> > I see. Then my production table should include a path ID, right?
> 
> Yeah, unless you treat it more like a cell and have M9-1, M9-2, and normalize DB further.

Well, I’d like to keep all data within the machine schema (like all production data in `m1.production`), even if we would have multiple rows per path within the same table. Therefore no `M9-1`. :wink:

MRIIOT, 2021-08-02:
> > So the end_time would come from the absence of previous alarm/message.
> 
> Yes, it’d work just like MTConnect. However, `fanuc-driver` does not send a packet yet with empty alarm/message topic when an alarm is cancelled/fixed/dismissed (how do you call it?).

Dismissed sounds good.
Alarms do work.  Messages might not.  I have to test my changes.

> > Should be a var we can pull out from Fanuc ladder. You want to try to find it?
> 
> Yes, but I need some clarification: is _Fanuc ladder_ a list in one of the manuals? Or where could I find it? 😉
> 

I would check with OEM to see if they can provide a PMC address (https://www.inventcom.net/fanuc-focas-library/pmc/pmc_rdpmcrng)

> Also: do you think it is machine-dependent?
>
>
> I am not sure what you mean by _another business system_, buy I presume you mean a system in our company used to manage orders and tracking labours. [...]

Yes.

> Well, I’d like to keep all data within the machine schema (like all production data in `m1.production`), even if we would have multiple rows per path within the same table. Therefore no `M9-1`. 😉

tukusejssirs, 2021-08-02:
> Alarms do work. Messages might not. I have to test my changes.

Well, in my tests of two alarms (`fanuc/{machine}/messages1`), each on different machine are successful (I get the alarm code and message), so I don’t really understand why you say they _might not work_. Still my tests are not that thorough yet.

> I would check with OEM to see if they can provide a PMC address (https://www.inventcom.net/fanuc-focas-library/pmc/pmc_rdpmcrng). Do you have electrical schematics from OEM?

Yes, I have electrical schematics for several Fanucs, but not all. Some of them are PDFs, other hard copies. As I said, we already use PLCs to gather signals from the light indicators, therefore if you suggest to consult the schematics in order to get the light indicator status, we already done that.

To be clear: I was (and am) sceptic about getting the light indicator status via FOCAS. You suggested contacting the OEM if they can provide a PMC address. My question is: can I get this PMC address from the manuals. I have found a manual from a Doosan Puma 3100 XLY (check Telegram if you can understand that mumble-jumble).

MRIIOT, 2021-08-02:
> Well, in my tests of two alarms (`fanuc/{machine}/messages1`), each on different machine are successful (I get the alarm code and message), so I don’t really understand why you say they _might not [...]

You said there is no message published when message is dismissed. #22 

> Yes, I have electrical schematics for several Fanucs, but not all. Some of them are PDFs, other hard copies. As I said, we already use PLCs to gather signals from the light indicators, therefore if [...]

Did you hook up a third party PLC to intercept the stack light?  What did you use?

> To be clear: I was (and am) sceptic about getting the light indicator status via FOCAS. You suggested contacting the OEM if they can provide a PMC address. My question is: can I get this PMC address [...]

Looks like `Y0003.0, Y0003.1, Y0003.2`.

tukusejssirs, 2021-08-03:
> You said there is no message published when message is dismissed. #22

Indeed, I said in [#4 (comment)](https://github.com/Ladder99/fanuc-driver/issues/4#issuecomment-888195509) (not in #22) that _I could not find the alarm code nor message in MQTT log_, however, I have already explained this error of mine in [#4 (another comment)](https://github.com/Ladder99/fanuc-driver/issues/4#issuecomment-888314505):

> [@tukusejssirs]
>
> Anyway, I wrote the quoted sentence before I actually found the alarm code and message (I could not find it yesterday, but today I did), but I forgot to remove it. Also I misplaced the correct sentence at the end of `stat_info with 2360 alarm` accordion, where you can clearly see the correct statement: _That said, now I get `2360` alarm code and message in `fanuc/{machine}/message1`._ Sorry. :wink:

All in all: all alarms I tested I could get via `fanuc-driver`. :wink:

---

> Did you hook up a third party PLC to intercept the stack light? What did you use?

Yes. We use PLCs from WAGO, specifically:
- [Wago 750-362](https://www.wago.com/global/i-o-systems/fieldbus-coupler-modbus-tcp/p/750-362);
- [4-channel module Wago 750-402](https://www.wago.com/mx-en/io-systems/4-channel-digital-input/p/750-402);
- [terminal/end module Wago 750-600](https://www.wago.com/global/i-o-systems/end-module/p/750-600).

We use Modbus to communicate with PLC. Of course, we use it occasionally for other sensors too, like voltage and temperature for glass-melting furnaces, when we cannot use the data from the sensors in the particular machine.

Anyway, a _stack light_? Is it how you call the light indicator / ‘semaphore’? :wink:

> Looks like `Y0003.0, Y0003.1, Y0003.2`. https://www.youtube.com/watch?v=RM1TkANSKXM

Do I need to configure anything in order to read those values? Could we gather them right away via FOCAS? If yes, how?

MRIIOT, 2021-08-03:
> > You said there is no message published when message is dismissed. #22
> 
> Indeed, I said in [#4 (comment)](https://github.com/Ladder99/fanuc-driver/issues/4#issuecomment-888195509) (not in #22) that _I could not find the alarm code nor message in MQTT log_, however, I have already explained this error of mine in [#4 (another comment)](https://github.com/Ladder99/fanuc-driver/issues/4#issuecomment-888314505):
> 
> > [@tukusejssirs]
> > Anyway, I wrote the quoted sentence before I actually found the alarm code and message (I could not find it yesterday, but today I did), but I forgot to remove it. Also I misplaced the correct sentence at the end of `stat_info with 2360 alarm` accordion, where you can clearly see the correct statement: _That said, now I get `2360` alarm code and message in `fanuc/{machine}/message1`._ Sorry. 😉
> 
> All in all: all alarms I tested I could get via `fanuc-driver`. 😉
> 

OK

> Yes. We use PLCs from WAGO, specifically:
>
> * [Wago 750-362](https://www.wago.com/global/i-o-systems/fieldbus-coupler-modbus-tcp/p/750-362);
> * [4-channel module Wago [...]

I recently used an [OPTO22 groov Rio](https://www.opto22.com/products/groov-rio).  The different I/O supported is nice.
Mazak also has a custom firmware for the [Moxa](https://www.moxa.com/en/products/industrial-edge-connectivity/controllers-and-ios/universal-controllers-and-i-os/iologik-e2200-series/iologik-e2210) units which outputs MTConnect SHDR natively.

> Anyway, a _stack light_? Is it how you call the light indicator / ‘semaphore’? 😉

Yeah, stack light, andon light, broken light :)

> Do I need to configure anything in order to read those values? Could we gather them right away via FOCAS? If yes, how?

Will need to add the proper Focas call.  This would be a good to try next.

tukusejssirs, 2021-08-03:
> Yeah, stack light, andon light, broken light :)

:question::question::question: I get that [_stack light_](https://en.wikipedia.org/wiki/Stack_light) means that the lights are stacked, but I don’t understand _broken light_ (the light ain’t broken; maybe b/c of the production being ‘broken’, paused) and _[andon](https://en.wikipedia.org/wiki/Andon_(manufacturing)) light_ (I presume that word _andon_ come from Japanese). Anyway, thanks for point me to the right jargon.

tukusejssirs, 2021-08-03:
> > So the end_time would come from the absence of previous alarm/message.
> 
> Yes, it’d work just like MTConnect. However, `fanuc-driver` does not send a packet yet with empty alarm/message topic when an alarm is cancelled/fixed/dismissed (how do you call it?).

I think that there are two approaches:
1. send a packet whenever there is a change (any one alarm is triggered or dismissed);
2. potentially more MTConnect-compatible way:
   - send a packet every x seconds when at least one alarm is triggered;
   - when no alarm is triggered, send no packet unless the an alarm was dismissed (then send an empty array once).

I like the former one much more because of effectivity and performance.

MRIIOT, 2021-08-06:
> Awaiting your call! 😉

[image]

[image]

MRIIOT, 2021-08-06:
> I think that there are two approaches:
>
> 1. send a packet whenever there is a change (any one alarm is triggered or dismissed);
> 2. potentially more MTConnect-compatible way:
>
>    * send a packet [...]

I like 1 as well, which is how it works now + when alarm is dismissed, empty array is sent only once.
Why do you think 2 is more MTC?  With MQTT retained flag, the last alarm will always be at broker.  If MTC MQTT adapter disconnects and reconnects, the last alarm will be published to MTC MQTT adapter.

MRIIOT, 2023-03-10:
Support for maintaining alarms/op-messages through is being added in 0.7.
Alarms will remain in output even when they are cleared/dismissed with is_triggered=false.

closed 2023-03-10 by MRIIOT