Ladder99/fanuc-driver #41: Unable to load shared library 'FW LIB32.dll' with Fanuc 310i

tukusejssirs, 2021-10-30:
I’ve acquired access to another Fanuc machine (Gildemeister Sprint42 Linear with Fanuc 310i). There are two differences:
- the server is CentOS 7;
- and Fanuc is 310i.

I’ve got the following error when running `bin/Debug/netcoreapp3.1/linux-x64/publish/fanuc --nlog nlog.config --config config.yaml`:

```js
Collector sweep failed at segment BEGIN. System.DllNotFoundException: Unable to load shared library 'FW
LIB32.dll' or one of its dependencies. In order to help diagnose loading problems, consider setting the LD_DEBUG environment variable: libFWLIB32.dll:
 cannot open shared object file: No such file or directory
   at l99.driver.fanuc.Focas.cnc_allclibhndl3(Object ip, UInt16 port, Int32 timeout, UInt16& FlibHndl)
   at l99.driver.fanuc.Platform.<Connect>b__7_0() in /root/fanuc-driver/fanuc/platform/Connect.cs:line 17
   at l99.driver.fanuc.Platform.<>c.<.ctor>b__35_0(Func`1 nativeCallWrapper) in /root/fanuc-driver/fanuc/platform/Platform.cs:line 31
   at l99.driver.fanuc.Platform.Connect() in /root/fanuc-driver/fanuc/platform/Connect.cs:line 15
   at l99.driver.fanuc.Platform.ConnectAsync() in /root/fanuc-driver/fanuc/platform/Connect.cs:line 10
   at CallSite.Target(Closure , CallSite , Object )
   at l99.driver.fanuc.collectors.FanucCollector2.CollectBeginAsync() in /root/fanuc-driver/fanuc/collectors/FanucCollector2.cs:line 497
   at l99.driver.fanuc.collectors.UseCase01.CollectBeginAsync() in /root/fanuc-driver/fanuc/collectors/UseCase01.cs:line 90
   at l99.driver.fanuc.collectors.FanucCollector2.CollectAsync() in /root/fanuc-driver/fanuc/collectors/FanucCollector2.cs:line 388
```

It seems to be similar to what I have encountered when I first tried `fanuc-driver` (see #4), but this time it does not work with both `l99.driver.fanuc.collectors.Basic01` and `l99.driver.fanuc.collectors.UseCase01`.

MRIIOT, 2021-11-09:
I would have to review the Focas docs, but I wonder if we have a situation where 310/150/180 might be Windows/Focas1 only?

tukusejssirs, 2021-11-09:
Well, somewhere on that controller I found that it is uses Winc#dows CE 4.0. As for FOCAS1, I have no idea.

closed 2022-02-01 by MRIIOT