Ladder99/fanuc-driver #42: Collector sweep failed at segment BEGIN. System.DllNotFoundException: Unable to load shared library 'FWLIB32.dll' or one of its dependencies.

tukusejssirs, 2021-11-12:
@MRIIOT, I get this error on 5f4abb9 (`main` branch). It didn’t happen with a version from August 2021 (I may try to indentify it) with the old `config.yml` which didn’t have `machines['l99.driver.fanuc.FanucMachine, fanuc']` object (but included its properties as part of `machines` object.

Below is the config I used, commands and the error I get when using this config is after it.

```yaml
---
machines:
  - id: m4
    enabled: !!bool true
    type: l99.driver.fanuc.FanucMachine, fanuc
    strategy: l99.driver.fanuc.collectors.UseCase01, fanuc
    handler: l99.driver.fanuc.handlers.Native, fanuc
    sweep_ms: !!int 1000
    net_ip: 10.14.7.8
    net_port: !!int 8193
    net_timeout_s: !!int 10
    broker:
      enabled: !!bool true
      net_ip: localhost
      net_port: !!int 1883
      auto_connect: !!bool true
      publish_status: !!bool true
      publish_arrivals: !!bool false
      publish_changes: !!bool true
      publish_disco: !!bool true
      disco_base_topic: fanuc
      anonymous: !!bool true
      user:
      password:

    l99.driver.fanuc.FanucMachine, fanuc:
      sweep_ms: !!int 1000
      net_ip: 10.14.7.8
      net_port: !!int 8193
      net_timeout_s: !!int 3
...
```

```bash
# All commands run in `fanuc/` subfolder of the repo

# I am not sure if I need to build the code prior to publishing it
DOTNET_CLI_TELEMETRY_OPTOUT='true' dotnet build /nowarn:CS0618 /nowarn:CS8632 /nowarn:CS1998 -p:DefineConstants=LINUX64 fanuc/fanuc.csproj

DOTNET_CLI_TELEMETRY_OPTOUT='true' dotnet publish fanuc/fanuc.csproj --self-contained true --runtime linux-x64 /nowarn:CS0618 /nowarn:CS8632 /nowarn:CS1998

bin/Debug/netcoreapp3.1/linux-x64/publish/fanuc --nlog nlog.config --config m4.yml
```

<details>
  <summary>error</summary>

```bash
Bitness: 64-bit
Argument '--nlog' = 'nlog.config'
Argument '--config' = 'm4_use_case_01.yaml'
2021/11/12 18:03:30.249|TRACE|Deserialized configuration:
{
  "machines": [
    {
      "id": "m4",
      "enabled": true,
      "type": "l99.driver.fanuc.FanucMachine, fanuc",
      "strategy": "l99.driver.fanuc.collectors.UseCase01, fanuc",
      "handler": "l99.driver.fanuc.handlers.Native, fanuc",
      "sweep_ms": 1000,
      "net_ip": "10.14.7.8",
      "net_port": 8193,
      "net_timeout_s": 10,
      "broker": {
[... 145 lines cut ...]
   at l99.driver.fanuc.Platform.Connect() in /home/ts/git/c10n/fanuc-driver/fanuc/platform/Connect.cs:line 15
   at l99.driver.fanuc.Platform.ConnectAsync() in /home/ts/git/c10n/fanuc-driver/fanuc/platform/Connect.cs:line 10
   at CallSite.Target(Closure , CallSite , Object )
   at System.Dynamic.UpdateDelegates.UpdateAndExecute1[T0,TRet](CallSite site, T0 arg0)
   at l99.driver.fanuc.collectors.FanucCollector2.CollectBeginAsync() in /home/ts/git/c10n/fanuc-driver/fanuc/collectors/FanucCollector2.cs:line 497
   at l99.driver.fanuc.collectors.UseCase01.CollectBeginAsync() in /home/ts/git/c10n/fanuc-driver/fanuc/collectors/UseCase01.cs:line 90
   at l99.driver.fanuc.collectors.FanucCollector2.CollectAsync() in /home/ts/git/c10n/fanuc-driver/fanuc/collectors/FanucCollector2.cs:line 388|l99.driver.fanuc.collectors.UseCase01|
2021/11/12 18:03:32.214|TRACE|1636736612213 'localhost:1883/m4' ARRIVE 179b => fanuc/m4-all/ping
{"observation":{"time":1636736612114,"machine":"m4","name":"ping"},"source":{"data":{"_id":"m4","IPAddress":"10.14.7.8","Port":8193,"ConnectionTimeout":3}},"state":{"data":"NOK"}} |l99.driver.base.mqtt.Broker|
2021/11/12 18:03:32.218|TRACE|1636736612218 'localhost:1883/m4' CHANGE 179b => fanuc/m4/ping
{"observation":{"time":1636736612114,"machine":"m4","name":"ping"},"source":{"data":{"_id":"m4","IPAddress":"10.14.7.8","Port":8193,"ConnectionTimeout":3}},"state":{"data":"NOK"}} |l99.driver.base.mqtt.Broker|
2021/11/12 18:03:32.267|TRACE|[m4] Sweep delay: 811ms |l99.driver.fanuc.collectors.UseCase01|
2021/11/12 18:03:33.084|ERROR|[m4] Collector sweep failed at segment BEGIN. System.DllNotFoundException: Unable to load shared library 'FWLIB32.dll' or one of its dependencies. In order to help diagnose loading problems, consider setting the LD_DEBUG environment variable: libFWLIB32.dll: cannot open shared object file: No such file or directory
   at l99.driver.fanuc.Focas.cnc_allclibhndl3(Object ip, UInt16 port, Int32 timeout, UInt16& FlibHndl)
   at l99.driver.fanuc.Platform.<Connect>b__7_0() in /home/ts/git/c10n/fanuc-driver/fanuc/platform/Connect.cs:line 17
   at l99.driver.fanuc.Platform.<>c.<.ctor>b__35_0(Func`1 nativeCallWrapper) in /home/ts/git/c10n/fanuc-driver/fanuc/platform/Platform.cs:line 31
   at l99.driver.fanuc.Platform.Connect() in /home/ts/git/c10n/fanuc-driver/fanuc/platform/Connect.cs:line 15
   at l99.driver.fanuc.Platform.ConnectAsync() in /home/ts/git/c10n/fanuc-driver/fanuc/platform/Connect.cs:line 10
   at CallSite.Target(Closure , CallSite , Object )
   at l99.driver.fanuc.collectors.FanucCollector2.CollectBeginAsync() in /home/ts/git/c10n/fanuc-driver/fanuc/collectors/FanucCollector2.cs:line 497
   at l99.driver.fanuc.collectors.UseCase01.CollectBeginAsync() in /home/ts/git/c10n/fanuc-driver/fanuc/collectors/UseCase01.cs:line 90
   at l99.driver.fanuc.collectors.FanucCollector2.CollectAsync() in /home/ts/git/c10n/fanuc-driver/fanuc/collectors/FanucCollector2.cs:line 388|l99.driver.fanuc.collectors.UseCase01|
```

</details>

MRIIOT, 2022-02-01:
We resolved this.  This was a combination of missing compiler definition and build+publish.

closed 2022-02-01 by MRIIOT

tukusejssirs, 2022-02-01:
Just for completeness, one needs to run `dotnet publish` command with `-p:DefineConstants=LINUX64`. `build` command is not necessary, but before rebuilding, one needs to remove `fanuc/{bin,obj}`, otherwise running `fanuc-driver` fails for some reason.

Updated commands:

```bash
# Remove older build
rm -rf fanuc/{bin,obj}

# Build and publish
DOTNET_CLI_TELEMETRY_OPTOUT='true' dotnet publish fanuc/fanuc.csproj --self-contained true --runtime linux-x64 /nowarn:CS0618 /nowarn:CS8632 /nowarn:CS1998 -p:DefineConstants=LINUX64

# Run
bin/Debug/netcoreapp3.1/linux-x64/publish/fanuc --nlog nlog.config --config m4.yml
```