Ladder99/fanuc-driver #45: Collector sweep failed at segment ROOT

tukusejssirs, 2022-01-18:
@MRIIOT, I get the following error once a second using `fanuc-driver` (606fce3 + `base-driver` c129ffc):

```bash
2022/01/18 19:53:22.145|ERROR|[m1] Collector sweep failed at segment ROOT. System.DllNotFoundException: Unable to load shared library 'FWLIB32.dll' or one of its dependencies. In order to help diagnose loading problems, consider setting the LD_DEBUG environment variable: libFWLIB32.dll: cannot open shared object file: No such file or directory
   at l99.driver.fanuc.Focas.pmc_rdpmctitle(UInt16 FlibHndl, ODBPMCTITLE a)
   at CallSite.Target(Closure , CallSite , Type , UInt16 , Object )
   at l99.driver.fanuc.Platform.<>c__DisplayClass90_0.<RdPmcTitle>b__0() in /home/ts/git/itens/otto/fanuc-driver/fanuc/platform/RdPmcTitle.cs:line 19
   at l99.driver.fanuc.Platform.<>c.<.ctor>b__35_0(Func`1 nativeCallWrapper) in /home/ts/git/itens/otto/fanuc-driver/fanuc/platform/Platform.cs:line 31
   at l99.driver.fanuc.Platform.RdPmcTitle() in /home/ts/git/itens/otto/fanuc-driver/fanuc/platform/RdPmcTitle.cs:line 17
   at l99.driver.fanuc.Platform.RdPmcTitleAsync() in /home/ts/git/itens/otto/fanuc-driver/fanuc/platform/RdPmcTitle.cs:line 10
   at CallSite.Target(Closure , CallSite , Object )
   at l99.driver.fanuc.collectors.UseCase01.CollectRootAsync() in /home/ts/git/itens/otto/fanuc-driver/fanuc/collectors/UseCase01.cs:line 116
   at l99.driver.fanuc.collectors.FanucCollector2.CollectAsync() in /home/ts/git/itens/otto/fanuc-driver/fanuc/collectors/FanucCollector2.cs:line 396|l99.driver.fanuc.collectors.UseCase01|
```

tukusejssirs, 2022-01-18:
I use the following `config.yml`:

```yml
---
machines:
  - id: m9
    enabled: !!bool true
    type: l99.driver.fanuc.FanucMachine, fanuc
    strategy: l99.driver.fanuc.collectors.UseCase01, fanuc
    handler: l99.driver.fanuc.handlers.Native, fanuc

    broker:
      enabled: !!bool true
      net_ip: localhost
      net_port: !!int 8193
      auto_connect: !!bool true
      publish_status: !!bool true
      publish_arrivals: !!bool false
      publish_changes: !!bool true
      publish_disco: !!bool true
      disco_base_topic: fanuc
      anonymous: !!bool true
      user: null
      password: null

    l99.driver.fanuc.FanucMachine, fanuc:
      sweep_ms: !!int 1000
      net_ip: 10.14.7.18
      net_port: !!int 8193
      net_timeout_s: !!int 10
...
```

tukusejssirs, 2022-01-18:
Can `machines[id].broker.net_port` and `machines[id].['l99.driver.fanuc.FanucMachine, fanuc'].net_port` have the same value (`8193` in this case)?

---

Another version version of the error:

```bash
2022/01/18 20:30:31.978|ERROR|[m9] Collector sweep failed at segment PATH. System.DllNotFoundException: Unable to load shared library 'FWLIB32.dll' or one of its dependenci
es. In order to help diagnose loading problems, consider setting the LD_DEBUG environment variable: libFWLIB32.dll: cannot open shared object file: No such file or director
y
   at l99.driver.fanuc.Focas.cnc_exeprgname(UInt16 FlibHndl, ODBEXEPRG a)
   at l99.driver.fanuc.Platform.<>c__DisplayClass17_0.<ExePrgName>b__0() in /home/ts/git/itens/otto/fanuc-driver/fanuc/platform/ExePrgName.cs:line 19
   at l99.driver.fanuc.Platform.<>c.<.ctor>b__35_0(Func`1 nativeCallWrapper) in /home/ts/git/itens/otto/fanuc-driver/fanuc/platform/Platform.cs:line 31
   at l99.driver.fanuc.Platform.ExePrgName() in /home/ts/git/itens/otto/fanuc-driver/fanuc/platform/ExePrgName.cs:line 17
   at l99.driver.fanuc.Platform.ExePrgNameAsync() in /home/ts/git/itens/otto/fanuc-driver/fanuc/platform/ExePrgName.cs:line 10
   at CallSite.Target(Closure , CallSite , Object )
   at System.Dynamic.UpdateDelegates.UpdateAndExecute1[T0,TRet](CallSite site, T0 arg0)
   at l99.driver.fanuc.collectors.ProgramAndPieceCount.CollectForEachPathAsync(Int16 current_path, Object path_marker) in /home/ts/git/itens/otto/fanuc-driver/fanuc/collectors/ProgramAndPieceCount.cs:line 50
   at System.Dynamic.UpdateDelegates.UpdateAndExecuteVoid1[T0](CallSite site, T0 arg0)
   at l99.driver.fanuc.collectors.FanucCollector2.CollectAsync() in /home/ts/git/itens/otto/fanuc-driver/fanuc/collectors/FanucCollector2.cs:line 424|l99.driver.fanuc.collectors.ProgramAndPieceCount|
```

tukusejssirs, 2022-01-18:
Another error:

```bash
2022/01/18 20:42:20.926|ERROR|[m9] Collector sweep failed at segment PATH. System.DllNotFoundException: Unable to load shared library 'FWLIB32.dll' or one of its dependencies. In order to help diagnose loading problems, consider setting the LD_DEBUG environment variable: libFWLIB32.dll: cannot open shared object file: No such file or directory
   at l99.driver.fanuc.Focas.cnc_exeprgname(UInt16 FlibHndl, ODBEXEPRG a)
   at l99.driver.fanuc.Platform.<>c__DisplayClass17_0.<ExePrgName>b__0() in /root/fanuc-driver/fanuc/platform/ExePrgName.cs:line 19                                        
   at l99.driver.fanuc.Platform.<>c.<.ctor>b__35_0(Func`1 nativeCallWrapper) in /root/fanuc-driver/fanuc/platform/Platform.cs:line 31                                      
   at l99.driver.fanuc.Platform.ExePrgName() in /root/fanuc-driver/fanuc/platform/ExePrgName.cs:line 17                                                                    
   at l99.driver.fanuc.Platform.ExePrgNameAsync() in /root/fanuc-driver/fanuc/platform/ExePrgName.cs:line 10                                                               
   at CallSite.Target(Closure , CallSite , Object )
   at System.Dynamic.UpdateDelegates.UpdateAndExecute1[T0,TRet](CallSite site, T0 arg0) in /_/src/System.Linq.Expressions/src/System/Dynamic/UpdateDelegates.Generated.cs:line 125
   at l99.driver.fanuc.collectors.ProgramAndPieceCount.CollectForEachPathAsync(Int16 current_path, Object path_marker) in /root/fanuc-driver/fanuc/collectors/ProgramAndPieceCount.cs:line 50
   at System.Dynamic.UpdateDelegates.UpdateAndExecuteVoid1[T0](CallSite site, T0 arg0) in /_/src/System.Linq.Expressions/src/System/Dynamic/UpdateDelegates.Generated.cs:line 1584
   at l99.driver.fanuc.collectors.FanucCollector2.CollectAsync() in /root/fanuc-driver/fanuc/collectors/FanucCollector2.cs:line 424|l99.driver.fanuc.collectors.ProgramAndPieceCount|
2022/01/18 20:42:21.585|ERROR|[m9] Collector sweep failed at segment PATH. System.ArgumentException: An item with the same key has already been added. Key: 1              
   at System.Collections.Generic.Dictionary`2.TryInsert(TKey key, TValue value, InsertionBehavior behavior) in /_/src/System.Private.CoreLib/shared/System/Collections/Generic/Dictionary.cs:line 527
   at System.Collections.Generic.Dictionary`2.Add(TKey key, TValue value) in /_/src/System.Private.CoreLib/shared/System/Collections/Generic/Dictionary.cs:line 233        
   at l99.driver.fanuc.IntermediateModelGenerator.AddPath(Int16 path) in /root/fanuc-driver/fanuc/utils/IntermediateModelGenerator.cs:line 142                             
   at l99.driver.fanuc.collectors.FanucCollector2.CollectAsync() in /root/fanuc-driver/fanuc/collectors/FanucCollector2.cs:line 409|l99.driver.fanuc.collectors.ProgramAndPieceCount|
2022/01/18 20:42:22.571|ERROR|[m9] Collector sweep failed at segment PATH. System.ArgumentException: An item with the same key has already been added. Key: 1              
   at System.Collections.Generic.Dictionary`2.TryInsert(TKey key, TValue value, InsertionBehavior behavior) in /_/src/System.Private.CoreLib/shared/System/Collections/Generic/Dictionary.cs:line 527
   at System.Collections.Generic.Dictionary`2.Add(TKey key, TValue value) in /_/src/System.Private.CoreLib/shared/System/Collections/Generic/Dictionary.cs:line 233        
   at l99.driver.fanuc.IntermediateModelGenerator.AddPath(Int16 path) in /root/fanuc-driver/fanuc/utils/IntermediateModelGenerator.cs:line 142                             
   at l99.driver.fanuc.collectors.FanucCollector2.CollectAsync() in /root/fanuc-driver/fanuc/collectors/FanucCollector2.cs:line 409|l99.driver.fanuc.collectors.ProgramAndPieceCount|
```

tukusejssirs, 2022-01-18:
Actually, it seems like I cannot make `fanuc-driver` work on Linux x64 after the config breakup (7cdbd59). On the other hand, I could not successfully build a7fa72e; and 1e7a3f9 had the `collector initialization failed` error.

However, I have a backup of a `fanuc/bin` folder (I have no idea from which commit it was built, but it is using the config before break up) which I can successfully run.

In the end, I wish I could run the latest `fanuc-driver` version without issues. I suspect it might be a hidden bug in `fanuc-driver` or my misunderstanding of the current config (after break up).

tukusejssirs, 2022-01-18:
As for the `Unable to load shared library 'FWLIB32.dll'` error: wouldn’t it be easier to always use lowercase filenames? You know, on Windows `FWLIB32.dll` and `fwlib32.dll` refers to the same file, but on Linux (as per the default config, which is case-sensitive), they refer to different files. That said, I suspect that in this case, this is just a related thing, not the cause.

tukusejssirs, 2022-01-19:
@MRIIOT, as a workaround, I want to try Ubuntu 16.04.6 LTS (last x86 Ubuntu version), however what is the difference between `LINUX32_100` and `LINUX32_105`?

MRIIOT, 2022-01-24:
```
/* read pmc title data */
        [DllImport("FWLIB32.dll", EntryPoint = "pmc_rdpmctitle")]
        public static extern short pmc_rdpmctitle(ushort FlibHndl, [Out, MarshalAs(UnmanagedType.LPStruct)] ODBPMCTITLE a);
```

```
public class ODBPMCTITLE
```

needs to be declared for LINUX64, struct needs to be made explicit for x64

closed 2022-02-01 by MRIIOT