Ladder99/fanuc-driver #46: Collector sweep failed at segment PATH ProgramAndPieceCount

enck19, 2022-01-26:
Hi i get this error using fanuc-driver:
|ERROR|[T221] Collector sweep failed at segment PATH. System.ArgumentException: An item with the same key has already been added. Key: 1
   at System.Collections.Generic.Dictionary`2.TryInsert(TKey key, TValue value, InsertionBehavior behavior)
   at System.Collections.Generic.Dictionary`2.Add(TKey key, TValue value)
   at l99.driver.fanuc.IntermediateModelGenerator.AddPath(Int16 path) in /root/fanuc-driver/fanuc/utils/IntermediateModelGenerator.cs:line 142
   at l99.driver.fanuc.collectors.FanucCollector2.CollectAsync() in /root/fanuc-driver/fanuc/collectors/FanucCollector2.cs:line 409|l99.driver.fanuc.collectors.ProgramAndPieceCount|

I need to read product data from a Series 32i, program-name, pieces-produced, pieces-produced-life

my config.yml
machines:
  - id: T221
    enabled: !!bool true
    type: l99.driver.fanuc.FanucMachine, fanuc
    **strategy: l99.driver.fanuc.collectors.ProgramAndPieceCount, fanuc
    handler: l99.driver.fanuc.handlers.Simple01, fanuc**
    broker:
      enabled: !!bool true
      net_ip: 192.168.1.253
      net_port: !!int 1883
      auto_connect: !!bool true
      publish_status: !!bool true
      publish_arrivals: !!bool true
      publish_changes: !!bool true
      publish_disco: !!bool true
      disco_base_topic: fanuc
      anonymous: !!bool false
      user: xxx
      password: xxx
    l99.driver.fanuc.FanucMachine, fanuc:
      sweep_ms: !!int 2000
      net_ip: 192.168.1.221
      net_port: !!int 8193
      net_timeout_s: !!int 10

Response from mqtt have empty values
[image]

changing handler to Native give me same error.

S.O.  Operating System: Debian GNU/Linux 10 (buster)  Kernel: Linux 4.19.0-18-amd64  Architecture: x86-64
Last fanuc driver version
Compile option dotnet build /nowarn:CS0618 /nowarn:CS8632 /nowarn:CS1998 -p:DefineConstants=LINUX64
Thanks

enck19, 2022-01-26:
with  this change on config.yml
**strategy: l99.driver.fanuc.collectors.UseCase01, fanuc
handler: l99.driver.fanuc.handlers.Native, fanuc**

i have:

Collector sweep failed at segment ROOT. System.DllNotFoundException: Unable to load shared library 'FWLIB32.dll' or one of its dependencies. In order to help diagnose loading problems, consider setting the LD_DEBUG environment variable: libFWLIB32.dll: cannot open shared object file: No such file or directory
   at l99.driver.fanuc.Focas.pmc_rdpmctitle(UInt16 FlibHndl, ODBPMCTITLE a)
   at CallSite.Target(Closure , CallSite , Type , UInt16 , Object )
   at l99.driver.fanuc.Platform.<>c__DisplayClass92_0.<RdPmcTitle>b__0() in /root/fanuc-driver/fanuc/platform/RdPmcTitle.cs:line 19
   at l99.driver.fanuc.Platform.<>c.<.ctor>b__36_0(Func`1 nativeCallWrapper) in /root/fanuc-driver/fanuc/platform/Platform.cs:line 36
   at l99.driver.fanuc.Platform.RdPmcTitle() in /root/fanuc-driver/fanuc/platform/RdPmcTitle.cs:line 17
   at l99.driver.fanuc.Platform.RdPmcTitleAsync() in /root/fanuc-driver/fanuc/platform/RdPmcTitle.cs:line 10
   at CallSite.Target(Closure , CallSite , Object )
   at l99.driver.fanuc.collectors.UseCase01.CollectRootAsync() in /root/fanuc-driver/fanuc/collectors/UseCase01.cs:line 116
   at l99.driver.fanuc.collectors.FanucCollector2.CollectAsync() in /root/fanuc-driver/fanuc/collectors/FanucCollector2.cs:line 396|l99.driver.fanuc.collectors.UseCase01|
2022/01/26 14:38:41.317|INFO|[message1] Veneer error invocation result:
{
  "method": "cnc_rdopmsg",
  "invocationMs": 19,
  "doc": "https://docs.ladder99.com/drivers/fanuc-driver/focas-api/misc/cnc_rdopmsg",
  "success": false,
  "rc": 9,
  "request": {
    "cnc_rdopmsg": {
      "type": 0,
      "length": 262
    }
  },
  "response": {
    "cnc_rdopmsg": {
      "opmsg": {
        "msg1": {
          "datano": 0,
          "type": 0,
          "char_num": 0,
          "data": ""
        },
        "msg2": {
          "datano": 0,
          "type": 0,
          "char_num": 0,
          "data": ""
        },
        "msg3": {
          "datano": 0,
          "type": 0,
          "char_num": 0,
          "data": ""
        },
        "msg4": {
          "datano": 0,
          "type": 0,
          "char_num": 0,
          "data": ""
        },
        "msg5": {
          "datano": 0,
          "type": 0,
          "char_num": 0,
          "data": ""
        }
      }
    }
  }
} |l99.driver.fanuc.veneers.OpMsgs|

MRIIOT, 2022-01-26:
> at l99.driver.fanuc.IntermediateModelGenerator.AddPath(Int16 path) in /root/fanuc-driver/fanuc/utils/IntermediateModelGenerator.cs:line 142

This exception is interesting.  I'll add logging around this today.  I would like to investigate this.

> at l99.driver.fanuc.Focas.pmc_rdpmctitle(UInt16 FlibHndl, ODBPMCTITLE a)

Yup, `pmc_rdpmctitle` method has not been defined for x64 yet.  Comment out:

https://github.com/Ladder99/fanuc-driver/blob/70384b09b484fc1c6f4d1e35f55055fc0894a4e5/fanuc/collectors/UseCase01.cs#L59

and

https://github.com/Ladder99/fanuc-driver/blob/70384b09b484fc1c6f4d1e35f55055fc0894a4e5/fanuc/collectors/UseCase01.cs#L116

enck19, 2022-01-26:
done but error about FWLIB32.dll
i need only program-name, pieces-produced, pieces-produced-life and alarm

Veneer error invocation result:
{
  "method": "cnc_rdopmsg",
  "invocationMs": 20,
  "doc": "https://docs.ladder99.com/drivers/fanuc-driver/focas-api/misc/cnc_rdopmsg",
  "success": false,
  "rc": 9,
  "request": {
    "cnc_rdopmsg": {
      "type": 0,
      "length": 262
    }
  },
  "response": {
    "cnc_rdopmsg": {
      "opmsg": {
        "msg1": {
          "datano": 0,
          "type": 0,
          "char_num": 0,
          "data": ""
        },
        "msg2": {
          "datano": 0,
          "type": 0,
          "char_num": 0,
          "data": ""
        },
        "msg3": {
          "datano": 0,
          "type": 0,
          "char_num": 0,
          "data": ""
        },
        "msg4": {
          "datano": 0,
          "type": 0,
          "char_num": 0,
          "data": ""
        },
        "msg5": {
          "datano": 0,
          "type": 0,
          "char_num": 0,
          "data": ""
        }
      }
    }
  }
} |l99.driver.fanuc.veneers.OpMsgs|
2022/01/26 16:50:06.565|ERROR|[T78] Collector sweep failed at segment ROOT. System.DllNotFoundException: Unable to load shared library 'FWLIB32.dll' or one of its dependencies. In order to help diagnose loading problems, consider setting the LD_DEBUG environment variable: libFWLIB32.dll: cannot open shared object file: No such file or directory
   at l99.driver.fanuc.Focas.pmc_rdpmctitle(UInt16 FlibHndl, ODBPMCTITLE a)
   at CallSite.Target(Closure , CallSite , Type , UInt16 , Object )
   at l99.driver.fanuc.Platform.<>c__DisplayClass92_0.<RdPmcTitle>b__0() in /root/fanuc-driver/fanuc/platform/RdPmcTitle.cs:line 19
   at l99.driver.fanuc.Platform.<>c.<.ctor>b__36_0(Func`1 nativeCallWrapper) in /root/fanuc-driver/fanuc/platform/Platform.cs:line 36
   at l99.driver.fanuc.Platform.RdPmcTitle() in /root/fanuc-driver/fanuc/platform/RdPmcTitle.cs:line 17
   at l99.driver.fanuc.Platform.RdPmcTitleAsync() in /root/fanuc-driver/fanuc/platform/RdPmcTitle.cs:line 10
   at CallSite.Target(Closure , CallSite , Object )
   at l99.driver.fanuc.collectors.UseCase01.CollectRootAsync() in /root/fanuc-driver/fanuc/collectors/UseCase01.cs:line 116
   at l99.driver.fanuc.collectors.FanucCollector2.CollectAsync() in /root/fanuc-driver/fanuc/collectors/FanucCollector2.cs:line 396|l99.driver.fanuc.collectors.UseCase01|

------------------------------------------
**i try trough linux amd64 container**
comment out also gcode_blocks part , error below about gcode_blocks
Veneer error invocation result:
{
  "method": "cnc_rdblkcount",
Veneer error invocation result:
{
  "method": "cnc_rdopmsg",

**Now i have this error:**
INFO|[message1] Veneer error invocation result:
{
  "method": "cnc_rdopmsg",
  "invocationMs": 30,
  "doc": "https://docs.ladder99.com/drivers/fanuc-driver/focas-api/misc/cnc_rdopmsg",
  "success": false,
  "rc": 9,
  "request": {
    "cnc_rdopmsg": {
      "type": 0,
      "length": 262
    }
  },
  "response": {
    "cnc_rdopmsg": {
      "opmsg": {
        "msg1": {
          "datano": 0,
          "type": 0,
          "char_num": 0,
          "data": ""
        },
        "msg2": {
          "datano": 0,
          "type": 0,
          "char_num": 0,
          "data": ""
        },
        "msg3": {
          "datano": 0,
          "type": 0,
          "char_num": 0,
          "data": ""
        },
        "msg4": {
          "datano": 0,
          "type": 0,
          "char_num": 0,
          "data": ""
        },
        "msg5": {
          "datano": 0,
          "type": 0,
          "char_num": 0,
          "data": ""
        }
      }
    }
  }
} |l99.driver.fanuc.veneers.OpMsgs|

**But some data appear**
[image]

Thanks

MRIIOT, 2022-01-26:
> "rc": 9,

EW_PARAM(9) | CNC parameter errorSeries 30i/31i/32i, 0i-D/F, PMi-ACNC parameter '11931#1' must be 0.

MRIIOT, 2022-01-26:
> at l99.driver.fanuc.IntermediateModelGenerator.AddPath(Int16 path) in /root/fanuc-driver/fanuc/utils/IntermediateModelGenerator.cs:line 142

Can you build the latest changes and try with :

strategy: l99.driver.fanuc.collectors.ProgramAndPieceCount, fanuc
handler: l99.driver.fanuc.handlers.Simple01, fanuc

And set nlog.config to TRACE?

enck19, 2022-01-27:
Build and run last in Dockerfile.LINUX64
Attach driver.log

[driver.log.tar.gz](https://github.com/Ladder99/fanuc-driver/files/7948263/driver.log.tar.gz)
[driver-internal.log.tar.gz](https://github.com/Ladder99/fanuc-driver/files/7948264/driver-internal.log.tar.gz)

MRIIOT, 2022-01-27:
I pushed a change that I believe will fix your issue.

at l99.driver.fanuc.Focas.cnc_exeprgname(UInt16 FlibHndl, ODBEXEPRG a)

Please pull and rebuild.

enck19, 2022-01-28:
> > "rc": 9,
> 
> EW_PARAM(9) | CNC parameter errorSeries 30i/31i/32i, 0i-D/F, PMi-ACNC parameter '11931#1' must be 0.

sorry but this is no clear, is a parameter to set on fanuc?

enck19, 2022-01-28:
> I pushed a change that I believe will fix your issue.
> 
> at l99.driver.fanuc.Focas.cnc_exeprgname(UInt16 FlibHndl, ODBEXEPRG a)
> 
> Please pull and rebuild.

test done, attach driver and driver-internal, some error are present on stdout.
i have the production data now.
[driver.log.tar.gz](https://github.com/Ladder99/fanuc-driver/files/7958246/driver.log.tar.gz)
[driver-internal.log.tar.gz](https://github.com/Ladder99/fanuc-driver/files/7958247/driver-internal.log.tar.gz)

[image]

error related to rc9

2022/01/28 11:19:10.377|TRACE|[T221] Platform invocation result:
{
  "method": "cnc_rdalmmsg",
  "invocationMs": 13,
  "doc": "https://docs.ladder99.com/drivers/fanuc-driver/focas-api/misc/cnc_rdalmmsg",
  "success": true,
  "rc": 0,
  "request": {
    "cnc_rdalmmsg": {
      "type": 5,
      "num": 10
    }
  },
  "response": {
    "cnc_rdalmmsg": {
      "num": 0,
      "almmsg": {
        "msg1": {
          "alm_no": 0,
          "type": 0,
          "axis": 0,
          "dummy": 0,
          "msg_len": 0,
          "alm_msg": ""
        },
        "msg2": {
          "alm_no": 0,
          "type": 0,
          "axis": 0,
          "dummy": 0,
          "msg_len": 0,
          "alm_msg": ""
        },
        "msg3": {
          "alm_no": 0,
          "type": 0,
          "axis": 0,
          "dummy": 0,
          "msg_len": 0,
          "alm_msg": ""
        },
        "msg4": {
          "alm_no": 0,
          "type": 0,
          "axis": 0,
          "dummy": 0,
          "msg_len": 0,
          "alm_msg": ""
        },
        "msg5": {
          "alm_no": 0,
          "type": 0,
          "axis": 0,
          "dummy": 0,
          "msg_len": 0,
          "alm_msg": ""
        },
        "msg6": {
          "alm_no": 0,
          "type": 0,
          "axis": 0,
          "dummy": 0,
          "msg_len": 0,
          "alm_msg": ""
        },
        "msg7": {
          "alm_no": 0,
          "type": 0,
          "axis": 0,
          "dummy": 0,
          "msg_len": 0,
          "alm_msg": ""
        },
        "msg8": {
          "alm_no": 0,
          "type": 0,
          "axis": 0,
          "dummy": 0,
          "msg_len": 0,
          "alm_msg": ""
        },
        "msg9": {
          "alm_no": 0,
          "type": 0,
          "axis": 0,
          "dummy": 0,
          "msg_len": 0,
          "alm_msg": ""
        },
        "msg10": {
          "alm_no": 0,
          "type": 0,
          "axis": 0,
          "dummy": 0,
          "msg_len": 0,
          "alm_msg": ""
        }
      }
    }
  }
} |l99.driver.fanuc.Platform|

MRIIOT, 2022-01-28:
> > > "rc": 9,
> 
> > 
> 
> > EW_PARAM(9) | CNC parameter errorSeries 30i/31i/32i, 0i-D/F, PMi-ACNC parameter '11931#1' must be 0.
> 
> 
> 
> sorry but this is no clear, is a parameter to set on fanuc?

Correct.  Lol, welcome to the world of Fanuc API where nothing is clear or easy :)

MRIIOT, 2022-01-28:
What OS did you build container for?

enck19, 2022-01-28:
> What OS did you build container for?

build container with **docker build -f Dockerfile.LINUX64 --tag=ladder99/fanuc-driver:latest .**
on same Debian 10, Docker version 20.10.12, build e91ed57

enck19, 2022-01-28:
> > > > "rc": 9,
> > 
> > 
> > > 
> > 
> > 
> > > EW_PARAM(9) | CNC parameter errorSeries 30i/31i/32i, 0i-D/F, PMi-ACNC parameter '11931#1' must be 0.
> > 
> > 
> > sorry but this is no clear, is a parameter to set on fanuc?
> 
> Correct. Lol, welcome to the world of Fanuc API where nothing is clear or easy :)

i need a fanuc bible but basically I'm a sysadmin 😉

closed 2022-02-01 by MRIIOT