Ladder99/fanuc-driver #49: Move all FOCAS library files in a separate folder

tukusejssirs, 2022-02-01:
IMHO all those DLL and SO files look like a clutter. What do you think about moving them into a separate folder? Or maybe two: one for DLL files, the other for SO files? Or maybe put those two folders into a single folder named `focas` or `library` or `lib` or whatever you’d like.

MRIIOT, 2022-02-02:
Need to test this across operating systems: https://docs.microsoft.com/en-us/windows/win32/dlls/dynamic-link-library-search-order?redirectedfrom=MSDN

tukusejssirs, 2022-02-02:
> Applications can control the location from which a DLL is loaded by specifying a full path

I would use a full path with the path to the `fanuc` binary as avariable. IMHO that should work on any platform (at least on Linux and Windows).

MRIIOT, 2023-09-05:
Non-functional and potentially breaking cross-platform support.  Removing milestone.