Ladder99/fanuc-driver #51: Document MQTT JSON objects published by `ThreeWay` veneer

tukusejssirs, 2022-02-03:
Related: #9, #28

---

This is a WIP. In this issue I describe the MQTT JSONs as TypeScript interfaces, however, later they could be described in a different way in the docs.

---

```typescript
// `fanuc/m${machine}/connect`, `fanuc/m${machine}-all/connect`
export interface FanucConnect {
  observation: {  // Observation metadata
    time: number, // Timestamp in ms when the observation data was measured
    machine: string,  // Machine ID
    name: 'connect',  // Observation name
    marker: Record<string, never>  // TODO: Can any payload have `observation.marker.path_no` or only some specific payloads? Can there be any other property in this object? How exactly can we describe this object?
  },
  state: {  // Observation data
    time: number,  // Time (duration) in ms since the `fanuc-driver` connected to the machine when the observation gathered the data. Since `fanuc-driver` connected to the machine for the first time or the last time?
    data: {  // Observation data itself
      success: boolean  // If `fanuc-driver` successfully connected to the machine
    }
  }
}

// `fanuc/m${machine}/paths`, `fanuc/m${machine}-all/paths`
export interface FanucPaths {
  observation: {
    time: number,
    machine: string,
    name: 'paths',
    marker: Record<string, never>  // TODO: Can this payload have `observation.marker.path_no`? Can there be any other property in this object? Is this object always empty in this payload (`Record<string, never>` in TypeScript actually means that it should be always empty)? If it is always empty, should it not be removed altogether (just like in `/ping` topic)?
  },
  state: {
    time: number,
    data: {
      path_no: number,  // Currently used path number
      maxpath_no: number  // Maximum number of paths used/supported by the particular machine
    }
  }
}

// `fanuc/m${machine}/axis_names/${path}, `fanuc/m${machine}-all/axis_names/${path} (`${path}` is required even on single-path machines)
export interface FanucAxisNames {
  observation: {
    time: number,
    machine: number,
    name: 'axis_names',
    marker: {
      path_no?: number  // Is this property optional or required in this payload?
    }
  },
  state: {
    time: number,
    data: {
      axes: FanucAxisName[]  // An array of axis names (both linear and rotary)
    }
  }
}

export interface FanucAxisName {
  name: string,  // Axis name (e.g. `X`)
  suff: string  // Axis name suffix (on Fanuc I have access to, it is an empty string, but on Mazaks I saw axis names like `C2`, therefore the suffix here in that case could be `2`)
}

// `fanuc/m${machine}/spindle_names/${path}`, `fanuc/m${machine}-all/spindle_names/${path}` (`${path}` is required even on single-path machines)
export interface FanucSpindleNames {
  observation: {
    time: number,
    machine: number,
    name: 'spindle_names',
    marker: {
      path_no?: number  // Is this property optional or required in this payload?
    }
  },
  state: {
    time: number,
    data: {
      spindles: FanucSpindleName[]  // An array of spindle names (both linear and rotary)
    }
  }
}

export interface FanucSpindleName {
  name: string,  // Spindle name (e.g. `S`)
  suff1: string  // Spindle name suffix (e.g. `1`)
  suff2: string  // Spindle name suffix (e.g. `1`; it is empty in machines I checked)
}

// `fanuc/m${machine}/sys-info/${path}`, `fanuc/m${machine}-all/sys-info/${path}` (`${path}` is required even on single-path machines)
export interface FanucSystemInfo {
  observation: {
    time: number,
    machine: number,
    name: 'sys-info',
    marker: {
      path_no?: number  // Is this property optional or required in this payload?
    }
  },
  state: {
    time: number,
    data: {
      loader_control: boolean,
      i_series: boolean,
      compound_machining: boolean,
      transfer_line: boolean,
      model: string,  // e.g. `MODEL B`
      model_code: number,  // e.g. `34`
      max_axis: 32,AbortControllercnc_type: string,  // e.g. `Series 32i`,
      cnc_type_code: string,  // e.g. `32`; Can this be a number only or a string too?
      mt_type: string,  // e.g. `Lathe`
      mt_type_code: string,  // e.g. `T`
      series: string,  // e.g. `G527`
      version: string,  // e.g. `16.0`; Can this be a number only or a string too?
      axes: number  // e.g. `6`; What is the difference between `max_axis` and `axes`? Does `max_axes` mean max axes that can be used on a particular machine and `axes` the number of axes actually available on the machine (it seems so)?
    }
  }
}

// `fanuc/m${machine}/state/${path}`, `fanuc/m${machine}-all/state/${path}` (`${path}` is required even on single-path machines)
export interface FanucState {
  observation: {
    time: number,
    machine: number,
    name: 'state',
    marker: {
      path_no?: number  // Is this property optional or required in this payload?
    }
  },
  state: {
    time: number,
    data: {
      aut: number,  // e.g. `1`
      run: number,  // e.g. `0`
      motion: number,  // e.g. `0`
      mstb: number,  // e.g. `0`
      emergency: number,  // e.g. `0`
      alarm: number,  // e.g. `0`
      timers: {
        poweron_min: number,  // Machine poweron time in minutes (since the controller was last turned on)
        operating_min: number,  // Machine operating time in minutes (since the machine started working on a piece / pieces last time)
        cutting_min: number  // Machine cutting time in minutes (??? since the machine started cutting?)
      },
      override: {  // Various overrides
        feed: number,  // Feedrate override in percents
        rapid: number,  // Rapid override in percents
        spindle: number  // Spindle override in percents
      },
      modal: {  // What is this for?
        m1: number,  // e.g. `5`
        m2: number,  // e.g. `0`
        m3: number,  // e.g. `0`
        t: number  // e.g. `707`
      }
    }
  }
}

// `fanuc/m${machine}/production/${path}`, `fanuc/m${machine}-all/production/${path}` (`${path}` is required even on single-path machines)
export interface FanucProduction {
  observation: {
    time: number,
    machine: number,
    name: 'production',
    marker: {
      path_no?: number  // Is this property optional or required in this payload?
    }
  },
  state: {
    time: number,
    data: {
      program: {
        name: string,  // e.g. `01590`; max length: 32 (or 31?) characters
        number: number,  // e.g. `1590`
        size_b: number,  // e.g. `1000`; ???
        comment: string,
        modified: number,  // Timestamp in ms
      },
      pieces: {
        produced: number,  // How many pieces were produced on a particular program (since last program load)
        produced_life: number,  // How many pieces were produced in the machine's lifetime (or Fanuc OS reinstall?)
        remaining: number  // How many pieces should be produced (set by the operator) before the program should stop producing any more pieces
      },
      timers: {
        cycle_time_ms: number  // How long (in ms) a piece (or last piece?) is/was produced
      }
    }
  }
}

// `fanuc/m${machine}/alarms/${path}`, `fanuc/m${machine}-all/alarms/${path}` (`${path}` is required even on single-path machines)
export interface FanucAlarms {
  observation: {
    time: number,
    machine: number,
    name: 'alarms',
    marker: {
      path_no?: number  // Is this property optional or required in this payload?
    }
  },
  state: {
    time: number,
    data: {
      alarms: []  // TODO: Update the array structure.
    }
  }
}

// `fanuc/m${machine}-all/focas_perf`
export interface FanucFocasPerf {
  observation: {
    time: number,
    machine: number,
    name: 'focas_perf',
    marker: Record<string, never>
  },
  state: {
    time: number,
    data: {
      sweepMs: number,
      invocation: {
        count: number,
        mxMethod: 'cnc_rdalmmsg_ALL',
        maxMs: number,
        minMs: number,
        avgMs: number,
        sumMs: number,
        failedMethods: FanucFocasFailedMethods[]
      }
    }
  }
}

export interface FanucFocasFailedMethods {
  method: string,  // e.g. `cnc_rdopmsg`; TODO: List all supported methods in an enum and use that of `method` property type
  rc: number
}

// `fanuc/m${machine}/ping`, `fanuc/m${machine}-all/ping`
export interface FanucPing {
  observation: {
    time: number,
    machine: number,
    name: 'ping'
  },
  state: {
    data: {
      data: 'OK' | 'NOK'  // TODO: Can there be an additional value?
    }
  }
}
```

tukusejssirs, 2022-03-16:
@MRIIOT, I have a couple of questions still unanswered:
- What does `[fanuc/m${machine}/production/${path}].state.data.timers.cycle_time_ms` actually mean? How long (in ms) last finished piece was produced? Or how long the current piece is being produced (it is not yet finished)? Or how long the program is being loaded?
- What does `[fanuc/m${machine}/production/${path}].state.data.program.size_b` actually mean?

MRIIOT, 2022-03-17:
> @MRIIOT, I have a couple of questions still unanswered:
> 
> * What does `[fanuc/m${machine}/production/${path}].state.data.timers.cycle_time_ms` actually mean? How long (in ms) last finished piece was produced? Or how long the current piece is being produced (it is not yet finished)? Or how long the program is being loaded?
>   
>   * **Answer:** Based on [`cnc_rdprogdir3` docs](https://www.inventcom.net/fanuc-focas-library/program/cnc_rdprogdir3): `buf` structure (object?) contains `mdate` which is an approximate time when the program file was last modified (`mtime`).
>   * @MRIIOT, as this parameter is invalid for `Series 15i`, will the MQTT message contain `modified` property? If yes, will it be set to `-1` (my assumption is based on the [source code](https://github.com/Ladder99/fanuc-driver/blob/develop/fanuc/veneers/ProductionData.cs#L21))?

Yes.

> * What does `[fanuc/m${machine}/production/${path}].state.data.program.size_b` actually mean?
>   
>   * **Answer:** Based on [`cnc_rdprogdir3` docs](https://www.inventcom.net/fanuc-focas-library/program/cnc_rdprogdir3): `buf` structure (object?) contains `length` which _program size (number of characters)_.
>   * I’d suggest to rename `size_b` to something a bit more human-friendly like `total_char_number`. I suggest to refrain from using `size` (as it is not file size, although it should be close to that) and `length` is a bit ambiguous too.

That would be fine.

closed 2023-03-10 by MRIIOT