Ladder99/fanuc-driver #57: Docker: Keep logs in a separate location from config files

tukusejssirs, 2022-03-15:
I have noticed though that you create `driver.log` in `/etc/fanuc`. Could we keep the logs separately from config?

We could use either of the following locations:
- `/etc/fanuc/config` for config and `/etc/fanuc/logs` for logs;
- `/etc/fanuc` for config and `/var/log/fanuc` for logs.

tukusejssirs, 2022-03-19:
Actually, it might be quite simple to solve this issue:

1. In `nlog.config`, we needs to substitute `/etc/fanuc` with (e.g.) `/var/log/fanuc`.
2. Bind some host path to `/var/log/fanuc`.

The only thing I am not sure whether the path `/etc/fanuc` for the logs is hardcoded somewhere in `fanuc-driver` or not.

MRIIOT, 2022-10-13:
All logging goes through NLog, so changes inside of `nlog.config` would affect both application and NLog internal logs.  And then another volume mapped to `/var/log/fanuc` would resolve this.