Ladder99/fanuc-driver #62: Implement a REST API

tukusejssirs, 2022-09-01:
Here is my suggestion how the REST API could look like.

We could also consider to add an endpoint to configure log path(s).

- `config.yml` would be used as initial configuration if that file exists (if not, don’t error out):
   - later, we might want to consider using a kind of configuration persistance (like into a database or even simply updating the `config.yml` file);
   - when the API is fully featured and functional, we should consider deprecating `config.yml` and use API requests to configure everything;
- I suggest to use JSON for both HTTP body and for returned value;
- notes:
   - in configuration objects, I use TypeScript types:
      - all property values are property data types;
      - when a question mark is placed before colon, that means that the property is optional;
- consider disabling management (CRUD operations) of `transport` and create some predefined tranforts that could be used by a machine:
   - reasoning: `fanuc-driver` only supports predefined transportation values, therefore there is no need to define/configure them by users;
   - `transport` in machine configuration could be configured to a string like `null`, `mqtt`, `influx`, `shdr`, `spb`;
   - TODO: where to put transport-related config like host or port?
- suggested API endpoints:
   - `GET /app/version`:
      - get `fanuc-driver` version;
      - something like this:
      
         ```js
         {
            commit: string,  // Git hash
            isDev: boolean,  // Whether the version is a dev version
            version: string  // Version number
         }
         ```

   - `PUT /app/exit`:
      - exit `fanuc-driver`;
      - this stops monitoring of all machines before actually exitting `fanuc-driver`;
   - `GET /machine`:
      - get configuration of all machines;
      - it would be an array of machine configuration objects;
   - `POST /machine`:
      - add a new machine to monitor;
      - a machine configuration object could look something like this:
      
         ```js
         {
            id: string,  // Unique machine ID
            enabled: boolean,  // Whether the monitoring of the machine should be enabled

            // TODO: add machine type property, e.g. `type`
            // - possible values would be, e.g., `fanuc`/`focas`, `mazak`/`mtconnect`;
            // - based on the value, `fanuc-driver` would choose appropriate driver/connector/protocol to gather the data from machine;
            type: string,  // `FanucMachine` (or `fanuc`); later even `MazakMachine` (or `mazak`)
            strategy: string,  // `FanucMultiStrategy` (or `multiStrategy`)
            handler: string,  // `fanucOne` (or `one`)

            changeFilter?: {
               fanucOne: {
                  changeOnly?: boolean,
                  skipInternal?: boolean
               }
            },

            noFilter?: {  // TODO: Is this actually needed? This is kind of duplicate of `changeFilter`.
               fanucOne: {
                  changeOnly?: boolean,
                  skipInternal?: boolean
               }
            },

            sources: source_id[],  // Array of target IDs

            // Array of collector names (they should be static and defined by `fanuc-driver`); e.g. `['ParameterDump', 'DiagnosisDump', 'PmcDump']` or `['MachineInfo', 'Alarms', 'Messages', 'StateData', 'ToolData', 'ProductionData', 'GCodeData', 'AxisData', 'SpindleData']`
            collectors: string[],

[... 99 lines cut ...]
      - a source configuration object could look something like this:
      
         ```ts
         {
            id: string,  // Unique source ID
            someName: string,  // For example: `FanucMachine`; I have no idea how to name this property
            sweep: number,  // In milliseconds
            net: {
               ip: string,
               port: number,
               timeout: number  // In seconds; consider changing this to milliseconds
            }
         }
         ```

   - `DELETE /source`:
      - delete all sources;
      - when a source is deleted, either automatically update all machines that use that particular source (i.e. remove the source from machine configuration) or simply reject the request to delete source with a list of machines that use a particular source (an array of sources with array of machines);
   - `GET /source/:source`:
      - get configuration of a source;
   - `PATCH /source/:source`:
      - modify configuration of a source;
   - `DELETE /source/:source`:
      - delete a targget;
      - when a source is deleted, either automatically update all machines that use that particular source (i.e. remove the source from machine configuration) or simply reject the request to delete source with a list of machines that use a particular source (an array of machines).

ottobolyos, 2024-01-12:
@MRIIOT, while REST API (with all endpoints specified in the issue description) would be preferred, how about adding some MQTT control topics to add and remove a machine to/from the monitored machines list at runtime?

I’d remove `machine.config.yaml` and and use `fanuc/add/:machineId` with the machine config as JSON payload and `fanuc-driver` would start monitoring the machine. Similarly, when I publish `fanuc/remove/:machineId` with no payload, `fanuc-driver` would stop monitoring the machine.

This way, no web server nor database would be required for this to accomplish.

WDYT? :thinking:

MRIIOT, 2024-01-13:
If anyone wants to sponsor this then of course.