Ladder99/fanuc-driver #65: Not able to run the Fanuc docker image

anilsagar, 2022-10-13:
Hi, 

I have followed the instructions of installing Fanuc docker image from [here ](https://docs.ladder99.com/en/latest/page/drivers/fanuc/installation-docker.html#clone-repository) but still getting an error that Fanuc container is not found. Could you pls suggest what could be the issue? 

Docker setup for ladder99 is already running in my machine.

[image]

tukusejssirs, 2022-10-13:
Did you build `fanuc-driver` (using `dotnet` v6+) and Docker container?

anilsagar, 2022-10-13:
Yes I used .net 6. Found out the issue. Profile has to be set individually for admin, fanuc, mqtt and influx. So the whole set of commands has to be run 4 times. Once for each of these 4 individual profiles. Thanks for your support.

closed 2022-10-13 by anilsagar