Ladder99/fanuc-driver #67: Unable to run fanuc driver

goutamkumar77, 2022-10-19:
Output:

```
gridx@GXVN00009:~/fanuc-driver/fanuc $ ./bin/Release/net6.0/fanuc --nlog ../../../../examples/fanuc-driver/nlog-example.config   --config ../../../../examples/fanuc-driver/config-example-makino.yml
Bitness: 64-bit
Argument '--nlog' = '../../../../examples/fanuc-driver/nlog-example.config'
Argument '--config' = '../../../../examples/fanuc-driver/config-example-makino.yml'
Unhandled exception. System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation.
 ---> System.DllNotFoundException: Unable to load shared library 'libfwlib32-linux-armv7.so.1.0.5' or one of its dependencies. In order to help diagnose loading problems, consider setting the LD_DEBUG environment variable: liblibfwlib32-linux-armv7.so.1.0.5: cannot open shared object file: No such file or directory
   at l99.driver.fanuc.Focas.cnc_startupprocess(Int64 level, String filename)
   at l99.driver.fanuc.Platform.<>c__DisplayClass134_0.<StartupProcess>b__0() in /home/gridx/fanuc-driver/fanuc/platform/StartupProcess.cs:line 16
   at l99.driver.fanuc.Platform.<>c.<.ctor>b__39_0(Func`1 nativeCallWrapper) in /home/gridx/fanuc-driver/fanuc/platform/Platform.cs:line 31
   at l99.driver.fanuc.Platform.StartupProcess(Int16 level, String filename) in /home/gridx/fanuc-driver/fanuc/platform/StartupProcess.cs:line 14
   at CallSite.Target(Closure , CallSite , Object , Int32 , String )
   at System.Dynamic.UpdateDelegates.UpdateAndExecuteVoid3[T0,T1,T2](CallSite site, T0 arg0, T1 arg1, T2 arg2)
   at l99.driver.fanuc.strategies.FanucStrategy..ctor(Machine machine, Object cfg) in /home/gridx/fanuc-driver/fanuc/strategies/FanucStrategy.cs:line 12
   at l99.driver.fanuc.strategies.FanucExtendedStrategy..ctor(Machine machine, Object cfg) in /home/gridx/fanuc-driver/fanuc/strategies/FanucExtendedStrategy.cs:line 38
   at l99.driver.fanuc.strategies.FanucMultiStrategy..ctor(Machine machine, Object cfg) in /home/gridx/fanuc-driver/fanuc/strategies/FanucMultiStrategy.cs:line 8
   --- End of inner exception stack trace ---
   at System.RuntimeMethodHandle.InvokeMethod(Object target, Span`1& arguments, Signature sig, Boolean constructor, Boolean wrapExceptions)
   at System.Reflection.RuntimeConstructorInfo.Invoke(BindingFlags invokeAttr, Binder binder, Object[] parameters, CultureInfo culture)
   at System.RuntimeType.CreateInstanceImpl(BindingFlags bindingAttr, Binder binder, Object[] args, CultureInfo culture)
   at System.Activator.CreateInstance(Type type, BindingFlags bindingAttr, Binder binder, Object[] args, CultureInfo culture, Object[] activationAttributes)
   at System.Activator.CreateInstance(Type type, Object[] args)
   at l99.driver.base.Machine.AddStrategyAsync(Type type, Object cfg) in /home/gridx/fanuc-driver/base-driver/base/Machine.cs:line 99
   at CallSite.Target(Closure , CallSite , Object )
   at System.Dynamic.UpdateDelegates.UpdateAndExecuteVoid1[T0](CallSite site, T0 arg0)
   at l99.driver.base.Machines.CreateMachines(Object config) in /home/gridx/fanuc-driver/base-driver/base/Machines.cs:line 138
   at CallSite.Target(Closure , CallSite , Object )
   at System.Dynamic.UpdateDelegates.UpdateAndExecute1[T0,TRet](CallSite site, T0 arg0)
   at l99.driver.fanuc.FanucService.ExecuteAsync(CancellationToken stoppingToken) in /home/gridx/fanuc-driver/fanuc/Program.cs:line 41
   at Microsoft.Extensions.Hosting.Internal.Host.StartAsync(CancellationToken cancellationToken)
   at Microsoft.Extensions.Hosting.HostingAbstractionsHostExtensions.RunAsync(IHost host, CancellationToken token)
   at Microsoft.Extensions.Hosting.HostingAbstractionsHostExtensions.RunAsync(IHost host, CancellationToken token)
   at Microsoft.Extensions.Hosting.HostingAbstractionsHostExtensions.Run(IHost host)
   at l99.driver.fanuc.Program.Main(String[] args) in /home/gridx/fanuc-driver/fanuc/Program.cs:line 19
   at l99.driver.fanuc.Program.<Main>(String[] args)
Aborted
```
I am unable to figure out this shared library thing.

MRIIOT, 2022-10-19:
@goutamkumar77 

What OS and architecture are you running on?  

Also, when you built the source, are the Focas libraries in `bin/Release/net6.0`  alongside the built binary?

MRIIOT, 2022-10-19:
@goutamkumar77 Oh, and if you are running ARM with 64bit, then you will either need to figure out how to link 32bit Focas libraries... or run as a container, built on top of 32bit base.

goutamkumar77, 2022-10-19:
OS: debian
Architecture: arm64

goutamkumar77, 2022-10-19:
> OS: debian Architecture: arm64

> @goutamkumar77 Oh, and if you are running ARM with 64bit, then you will either need to figure out how to link 32bit Focas libraries... or run as a container, built on top of 32bit base. There are no 64bit ARM Focas libraries, only 32bit.

Got it. Thanks. Will try.

MRIIOT, 2022-10-20:
Actually, what happens if you build against .NET 6 Arm32?

closed 2022-11-03 by MRIIOT

836138981, 2024-07-03:
> > OS: debian Architecture: arm64
> 
> > @goutamkumar77 Oh, and if you are running ARM with 64bit, then you will either need to figure out how to link 32bit Focas libraries... or run as a container, built on top of 32bit base. There are no 64bit ARM Focas libraries, only 32bit.
> 
> Got it. Thanks. Will try.

hello,this problem how to solve