Ladder99/fanuc-driver #73: Share single CNC connection across multiple output types.

MRIIOT, 2022-11-02:
Allow output to SHDR and MQTT, for example, from a single CNC connection to avoid multiple Focas connections to CNC.

tukusejssirs, 2022-11-02:
@MRIIOT, am I right to assume that this issue is about using a single FOCAS connection to a machine for every type of data gathered from that machine?

I would be awesome and a huge performance gain if we currently use a connection per feature (or something along the lines).

MRIIOT, 2022-11-02:
It is actually the other way around.  We observed performance degradation when using multiple Focas connections.  One to mqtt, one to Influx.

So the suggestion is to output to mqtt and Influx from a single Focas connection.

Even though the driver is async, I would imagine that the Focas connection blocks another somewhere.  But this should be Wiresharked.

MRIIOT, 2022-11-03:
@tukusejssirs, I don't think multiple connections to the same CNC will improve performance... From Focas docs:

> Multi-thread
> Win32 supports multi-thread. Each function in this library can be called from the multiple threads at a time. The **access to the single system resource, such as CNC/PMC's internal memory, at a time is exclusively controlled by this library**.

MRIIOT, 2023-01-28:
It is also worth noting that using Fast Ethernet instead of Embedded Ethernet improves performance.

tukusejssirs, 2023-01-28:
> It is also worth noting that using Fast Ethernet instead of Embedded Ethernet improves performance.

Is that a config on the machine or something you need to implement in `fanuc-driver`. If the former, how can you configure the machine to use fast eth?

MRIIOT, 2023-01-28:
It is a physically separate network interface on the machine (if available).