Ladder99/fanuc-driver #75: Maintain Focas connection handle across polling intervals, instead of re-connecting every polling interval.

MRIIOT, 2022-11-03:
Handle unmanaged memory cleanup and re-connection.

MRIIOT, 2023-01-31:
Synchonization context needs to be created in order to maintain Focas handle on a single thread.

MRIIOT, 2023-02-02:
l99.driver.fanuc.strategies.FanucMultiStrategy, fanuc:
      stay_connected: !!bool false

closed 2023-02-02 by MRIIOT

tukusejssirs, 2023-05-04:
@MRIIOT, what does this `stay_connected` config does? Does it mean whether `fanuc-driver` should stay connected to a machine via FOCAS or should it disconnect after each FOCAS call?

I see that in the example config, you set it to `false`. When should we set it to `true`?

MRIIOT, 2023-05-04:
`stay_connected=true` maintains the connection to the CNC until it is forced to disconnect (eg. network outage, CNC shut off) and then it will attempt to re-connect.  When set to `false`, it will connect, collect data, and disconnect, then repeat the process.

Example configuration should be changed to `true` as the default.