Ladder99/fanuc-driver #78: Constructor on type l99.driver.fanuc.FanucMachine not found

elliskot, 2022-12-06:
Hi, when start service I get this error: **Constructor on type 'l99.driver.fanuc.FanucMachine' not found**. I don't understand where is the problem.
Fanuc-driver is last commit (de0e822a) on main branch, and this is the my config:

thanks
Ellis

MRIIOT, 2022-12-21:
Sorry for the delay @elliskot .

Are you running it from source?

What happens when you run it without modifying the config (using one from examples folder as-is)?

MRIIOT, 2023-01-26:
Need more information.

closed 2023-01-26 by MRIIOT

elliskot, 2023-01-26:
Sorry I've missed your answer notification. (maybe it's finished in spam folder)
I used another library to do it.
I don't remember the error message, but the config in the source progect was wrong as it was pushed not correctly merged.

Ellis

MRIIOT, 2023-01-26:
@elliskot What library did you use?