Ladder99/fanuc-driver #80: Collectors need to be in particular order

tukusejssirs, 2023-01-13:
I use `ladder99/fanuc-driver:linux64-886a5a3` Docker image.

When I define collectors as follows (I sort the collectors alphabetically):

```yml
  collector-data: &collector-data
    l99.driver.fanuc.strategies.FanucMultiStrategy, fanuc:
      - l99.driver.fanuc.collectors.Alarms, fanuc
      - l99.driver.fanuc.collectors.AxisData, fanuc
      - l99.driver.fanuc.collectors.GCodeData, fanuc
      - l99.driver.fanuc.collectors.MachineInfo, fanuc
      - l99.driver.fanuc.collectors.Messages, fanuc
      - l99.driver.fanuc.collectors.ProductionData, fanuc
      - l99.driver.fanuc.collectors.SpindleData, fanuc
      - l99.driver.fanuc.collectors.StateData, fanuc
      - l99.driver.fanuc.collectors.ToolData, fanuc
```

it sometimes (rarely) works, sometimes it does not work.

However, when I move `MachineInfo` to the top (as the first collector, before `Alarms`), all works as expected.

I’d assume that the order in which the collectors are defined, does not matter, i.e. `fanuc-driver` would solve the collector dependencies on each other.

MRIIOT, 2023-01-28:
Yes, this needs to be addressed internally to resolve dependencies regardless of the order in yaml.

MRIIOT, 2023-09-05:
Order of collectors becomes relevant in the case of outputs backed by Scriban scripts.  In the example of MTConnect output, the `execution` data item is created by the `StateData` class, however, it might be desirable to overwrite the value using data from `Pmc` class.  In this case, `Pmc` must execute after `StateData`.