Ladder99/fanuc-driver #86: Influxdb Production measurement missing

Juanma2704, 2023-02-16:
Hello

I connected the fanuc-driver to Influxdb (v2.6). It worked perfectly untill a couple days ago when the all production data was missing while I was exploring the measurements and fields in Influx. However, it sends the data correctly to my mqtt broker running in localhost with mosquitto. I tried reinstalling the driver and downgrading my influxdb version but nothing seems to work. Would I have to change the influx transformers in order to solve it? I haven´t changed the system.xml file yet. Thank you.

MRIIOT, 2023-02-28:
```
production,machine=example-influx,path=1 program_name_selected="O2",program_comment_selected="(17137 CPO TOB BSP OP.B)",program_modified_selected=2023-02-16T15:22:00.0000000-03:00,program_name_current="O2",program_comment_current="(17137 CPO TOB BSP OP.B)",program_modified_current=2023-02-16T15:22:00.0000000-03:00,pieces_produced=41,pieces_produced_life=42,pieces_remaining=0,cycle_time=215312
```

I wonder if the above format does not comply with [Line Protocol ](https://docs.influxdata.com/influxdb/cloud/reference/syntax/line-protocol/)spec in some way.

@Juanma2704 can you test inserting the line using [INSERT](https://docs.influxdata.com/influxdb/v1.8/tools/shell/#write-data-to-influxdb-with-insert) and see if it fails?

MRIIOT, 2023-03-10:
If the datetime is an issue, you can try enclosing it in quotes or removing from the transformer.

```
l99.driver.fanuc.veneers.ProductionData, fanuc: >-
        {{observation.name}},
        {{-}}machine={{observation.machine}},
        {{-}}path={{observation.marker[0].number}}
        program_name_selected="{{data.program.selected.name}}",
        {{-}}program_comment_selected="{{data.program.selected.comment}}",
        {{-}}program_modified_selected="{{data.program.selected.modified}}",
        {{-}}program_name_current="{{data.program.current.name}}",
        {{-}}program_comment_current="{{data.program.current.comment}}",
        {{-}}program_modified_current="{{data.program.current.modified}}",
        {{-}}pieces_produced={{data.pieces.produced}},
        {{-}}pieces_produced_life={{data.pieces.produced_life}},
        {{-}}pieces_remaining={{data.pieces.remaining}},
        {{-}}cycle_time={{data.timers.cycle_time_ms}}
```

closed 2023-03-10 by MRIIOT