Ladder99/fanuc-driver #88: Configure veneers or collectors to gather only the data I need

Juanma2704, 2023-03-16:
Hi!

I was running the driver in Ubuntu and sending the data to Influxdb. In influx I get some issues with the duplicated values that for example state measurements sends, for example the ACTIVE value in the execution field is repeated five or six times when it should be showing only one value, tracing only the changes between the states. I believe this is because it sends values every time the poweron and cutting min changes. So I was wondering if there was a posibility of deleteing the cutting, power, operating min values from the veneer or collector.

MRIIOT, 2023-03-23:
@Juanma2704 you are correct, if a single property of the object changes, the entire object is sent over to Influx.  The simplest way I can think of is to add a C# method that could diff previous value and current value, and then generate the Influx line protocol from the difference.

Why is the current behavior a problem?

Juanma2704, 2023-03-23:
Thanks. I will try that. I had some issues with the influx database in grafana. Where I couldn´t calculate the stateDuration properly, anyway I solved it with othe influx function. By the way. Do you know another way of creating a Dashboard to visualize the data? Because I found out that I can´t embedd my grafana dashboards in my web page so I have to find an alternative.

MRIIOT, 2023-03-29:
@Juanma2704 I believe you can embed using Grafana Image server.

closed 2023-06-14 by MRIIOT