Ladder99/fanuc-driver #91: Could the log timestamps be in ISO 8601 format?

tukusejssirs, 2023-05-04:
Currently, the timestamp format is `2023/05/04 11:14:12.836` which is already in UTC (at least in my case), however, it is not obvious.

How about outputting it in ISO 8601 format? In this case, I don’t care if the timezone is UTC or local, but it should be obvious what is the timezone via outputting the timezone offset, e.g. `2023-05-04T11:14:12.836+00:00` or `2023-05-04T09:14:12.836+02:00`.

MRIIOT, 2023-09-05:
Yes, I can add a global flag in config to control the timestamp format.

MRIIOT, 2023-09-07:
https://github.com/Ladder99/fanuc-driver/blob/5dd464f1baef4ef9d6b86616267f389ecf864704/examples/windows/config.machines.yml#L130-L132

closed 2023-09-07 by MRIIOT

tukusejssirs, 2023-09-08:
Thanks for adding this. However, isn’t it vhanging the timestampt formation in the payload/data? This issue was about the _log_ timestamp format.

MRIIOT, 2023-09-08:
Oh, well in that case, this is configurable via nlog.config.