Ladder99/fanuc-driver #97: Program comment must be like O0001 in order to be recollected by the fanuc driver

Gafy27, 2023-07-21:
Hi,
I am working with a HWACHEON CUTEX 160B with a Fanuc 01 controller. The machine is connected to ethernet and sending all the data through the fanuc driver and all is working nicely. The problem I encounter is that the program comment is not reocllected if i dont use the nomenclature O0001 for each program, the programs now are named like this: D-11370-14-CN6-T1 and in the control is showed like this <D-11370-14-CN6-T1>.

We have another lathe that has the same controller and the program name is showed correctly, why this is happening?
Is it possible to adapt the driver to collect the program name in order to show the program comment in our system?
How the software search for the program name?

MRIIOT, 2023-09-05:
@Gafy27 sorry for the late reply.  

Below are the Focas API calls:

- Program number: https://www.inventcom.net/fanuc-focas-library/program/cnc_rdprgnum
- Program comment: https://www.inventcom.net/fanuc-focas-library/program/cnc_rdprogdir3

For program name (supported on series 0, 30, 31) you can try setting `unsupported` to `true` here: https://github.com/Ladder99/fanuc-driver/blob/9c3d575b62c0e481173401e65d92d7cb129437d8/examples/docker/config.machines.yml#L169-L171

- Program name: https://www.inventcom.net/fanuc-focas-library/program/cnc_exeprgname

Sometimes, you will need a custom `Collector` class.  Here, for example, the programs were hosted on C drive of the Windows controller.  So we had to extract data from the NC file on disk when the program number changed.  https://github.com/Ladder99/fanuc-driver/blob/9c3d575b62c0e481173401e65d92d7cb129437d8/examples/docker/config.machines.yml#L180C2-L201

MRIIOT, 2023-12-14:
Stale.

closed 2023-12-14 by MRIIOT