Ladder99/fanuc-driver #98: Ladder99/fanuc-driver - framework tried is .net core 3.1( x86). faced issue- got the error Unexpected Runtimeid 'win10x86'.

ajenita, 2023-08-21:
Please suggest me to run the code

MRIIOT, 2023-08-21:
Can you please attach logs or screenshots of the issue?

ottobolyos, 2023-08-21:
I presume the issue is related to the fact that `fanuc-driver` recently upgraded its .NET to v7 (see [this commit](https://github.com/Ladder99/fanuc-driver/commit/9e55095bdacee7eb3129dcc5a0d078d6e3ceaf4d)).

ajenita, 2023-08-22:
With dotnet runtime and sdk (version 7.0) I got the same issue unexpected runtimeid "win10x86"

MRIIOT, 2023-09-05:
Need more information (logs, screenshots, etc).  If you can provide this information, then reopen the issue.

closed 2023-09-05 by MRIIOT

ajenita, 2023-09-11:
[image]

Attached screenshot is my issue while building code in visual studio code. please provide suggestion to overcome this.

MRIIOT, 2023-09-12:
@ajenita This sounds like a VS Code issue.  I really do not know.