Ladder99/fanuc-driver #103: Fanuc 31i B - Shows Spindle 1 Production, State Data only on Driver Startup

ezioauditore0101, 2023-11-13:
Hello!
I am connecting to a DMG Mori machine powered with Fanuc 31i Model B with 3 spindles.
When I connected for the first time, all data was coming in rightly.
Without any changes, after driver restart I am getting state, production, gcode, alarms, messages only for Spindle 2. Spindle 1 is triggered when I start the driver (information is correct).
No updates were done to config.yml

MRIIOT, 2023-11-14:
Hi.  Can you turn up verbose logging in nlog.config and send the log?

ezioauditore0101, 2023-11-16:
--CONFIG FILE PATCH
- id: dmg_multispindle
    ## enabled: enable or disable collection for this machine
    enabled: !!bool true
    ## type: machine type
    type: l99.driver.fanuc.FanucMachine, fanuc
    ## strategy: strategy type
    strategy: l99.driver.fanuc.strategies.FanucMultiStrategy, fanuc
    ## handler: handler type
    handler: l99.driver.fanuc.handlers.FanucOne, fanuc
    ## transport: transport type
    transport: l99.driver.fanuc.transports.MQTT, fanuc
    ##
    ## machine instance attributes
    ##
    l99.driver.fanuc.FanucMachine, fanuc:
      ## sweep_ms: the interval at which machine is polled
      sweep_ms: !!int 1000
      net:
        ## ip: machine network address
        ip: 192.168.2.10
        ## port: focas port
        port: !!int 8193
        ## timeout_s: focas api timeout, must be above zero
        timeout_s: !!int 3
    ##
    ## strategy instance attributes
    ##
    l99.driver.fanuc.strategies.FanucMultiStrategy, fanuc:
      ## stay_connected: connect and disconnect every polling interval
      ##   this is helpful when collecting a lot of data to reduce the time we spend talking to the NC
      stay_connected: !!bool true
      ## exclusions: exclude paths, axes, and spindles from collection
      #exclusions:
      ## exclude path data only, but still collect axis and spindle data for this path
      #  1: 
      ## exlude axis X and spindle S data by name
      #  2: [ X, S ]
      ## exclude path data and all axes and spindles 
      #  3: [ % ]
      ## collectors: list of data collected by category
      collectors:
      - l99.driver.fanuc.collectors.MachineInfo, fanuc
      - l99.driver.fanuc.collectors.Alarms, fanuc
      - l99.driver.fanuc.collectors.Messages, fanuc
      - l99.driver.fanuc.collectors.StateData, fanuc
      - l99.driver.fanuc.collectors.ToolData, fanuc
      - l99.driver.fanuc.collectors.ProductionData, fanuc
      #- l99.driver.fanuc.collectors.ProductionDataExternalSubprogramDetails, fanuc
      - l99.driver.fanuc.collectors.GCodeData, fanuc
      - l99.driver.fanuc.collectors.AxisData, fanuc
      - l99.driver.fanuc.collectors.SpindleData, fanuc
    ##
    ## handler instance attributes
    ##
    l99.driver.fanuc.handlers.FanucOne, fanuc:
      ## change_only: send data only when it changes
      change_only: !!bool true
      ## skip_interval: send internal data, useful for troubleshooting
      skip_internal: !!bool true
    ##
    ## transport instance attributes
    ##
    l99.driver.fanuc.transports.MQTT, fanuc:
      topic: fanuc/{{machine.Id}}/{{veneer.Name}}{{if veneer.SliceKey}}/{{veneer.SliceKey}}{{end}}
      net:
        ## type: mqtt connection type, tcp or ws
        type: tcp
        ## ip: broker network address
        ip: MQTT_IP
[... 469 lines cut ...]
2023-11-16 10:41:28.5372 Debug Logger l99.driver.fanuc.collectors.GCodeData [Info] => fileTarget consoleTarget
2023-11-16 10:41:28.5372 Debug Logger l99.driver.fanuc.collectors.GCodeData [Warn] => fileTarget consoleTarget
2023-11-16 10:41:28.5372 Debug Logger l99.driver.fanuc.collectors.GCodeData [Error] => fileTarget consoleTarget
2023-11-16 10:41:28.5372 Debug Logger l99.driver.fanuc.collectors.GCodeData [Fatal] => fileTarget consoleTarget
2023-11-16 10:41:28.5372 Warn Logger: l99.driver.fanuc.collectors.AxisData configured with duplicate output to target: FileTarget(Name=fileTarget). LoggingRule with NamePattern='l99.driver.fanuc.*' and Level=Debug has been skipped.
2023-11-16 10:41:28.5372 Debug Targets configured when LogLevel >= Debug for Logger: l99.driver.fanuc.collectors.AxisData
2023-11-16 10:41:28.5508 Debug Logger l99.driver.fanuc.collectors.AxisData [Debug] => fileTarget consoleTarget
2023-11-16 10:41:28.5508 Debug Logger l99.driver.fanuc.collectors.AxisData [Info] => fileTarget consoleTarget
2023-11-16 10:41:28.5508 Debug Logger l99.driver.fanuc.collectors.AxisData [Warn] => fileTarget consoleTarget
2023-11-16 10:41:28.5508 Debug Logger l99.driver.fanuc.collectors.AxisData [Error] => fileTarget consoleTarget
2023-11-16 10:41:28.5508 Debug Logger l99.driver.fanuc.collectors.AxisData [Fatal] => fileTarget consoleTarget
2023-11-16 10:41:28.5508 Warn Logger: l99.driver.fanuc.collectors.SpindleData configured with duplicate output to target: FileTarget(Name=fileTarget). LoggingRule with NamePattern='l99.driver.fanuc.*' and Level=Debug has been skipped.
2023-11-16 10:41:28.5508 Debug Targets configured when LogLevel >= Debug for Logger: l99.driver.fanuc.collectors.SpindleData
2023-11-16 10:41:28.5508 Debug Logger l99.driver.fanuc.collectors.SpindleData [Debug] => fileTarget consoleTarget
2023-11-16 10:41:28.5508 Debug Logger l99.driver.fanuc.collectors.SpindleData [Info] => fileTarget consoleTarget
2023-11-16 10:41:28.5508 Debug Logger l99.driver.fanuc.collectors.SpindleData [Warn] => fileTarget consoleTarget
2023-11-16 10:41:28.5508 Debug Logger l99.driver.fanuc.collectors.SpindleData [Error] => fileTarget consoleTarget
2023-11-16 10:41:28.5508 Debug Logger l99.driver.fanuc.collectors.SpindleData [Fatal] => fileTarget consoleTarget
2023-11-16 10:41:28.5675 Warn Logger: l99.driver.fanuc.handlers.FanucOne configured with duplicate output to target: FileTarget(Name=fileTarget). LoggingRule with NamePattern='l99.driver.fanuc.*' and Level=Debug has been skipped.
2023-11-16 10:41:28.5675 Debug Targets configured when LogLevel >= Debug for Logger: l99.driver.fanuc.handlers.FanucOne
2023-11-16 10:41:28.5675 Debug Logger l99.driver.fanuc.handlers.FanucOne [Debug] => fileTarget consoleTarget
2023-11-16 10:41:28.5675 Debug Logger l99.driver.fanuc.handlers.FanucOne [Info] => fileTarget consoleTarget
2023-11-16 10:41:28.5675 Debug Logger l99.driver.fanuc.handlers.FanucOne [Warn] => fileTarget consoleTarget
2023-11-16 10:41:28.5675 Debug Logger l99.driver.fanuc.handlers.FanucOne [Error] => fileTarget consoleTarget
2023-11-16 10:41:28.5675 Debug Logger l99.driver.fanuc.handlers.FanucOne [Fatal] => fileTarget consoleTarget

MRIIOT, 2023-11-17:
@ezioauditore0101 the logs I am looking for are `driver.2023-11-17.log`.  Do you see those?

ezioauditore0101, 2023-11-20:
attaching the file.
The program was already running so I restarted the driver and hence immediately you will see GCode entries.

Driver restarted on 2023-11-20 10:48:10 -> DMG/state/1 received on 2023-11-20 10:48:23 ; not after. 
Same applicable for /tool/1 /messages/1 /production/1 etc.
[driver.2023-11-20-after-restart.txt](https://github.com/Ladder99/fanuc-driver/files/13406615/driver.2023-11-20-after-restart.txt)

MRIIOT, 2023-11-21:
All I can tell from the logs is that reading active program pointer is failing with error 13.  

```
CNC execution denied
- No program is executed.
- CNC is in DNC/M198 or RISC operation.
- STL signal is ON.(FS30i/31i/32i, 0i-D/F, PMi-A)
- CNC is in the emergency stop.(FS30i/31i/32i, 0i-D/F, PMi-A)
```
 
I would suggest commenting out as many collectors as you can to isolate the issue.

```
collectors:
      - l99.driver.fanuc.collectors.MachineInfo, fanuc
      #- l99.driver.fanuc.collectors.Alarms, fanuc
      #- l99.driver.fanuc.collectors.Messages, fanuc
      - l99.driver.fanuc.collectors.StateData, fanuc
      #- l99.driver.fanuc.collectors.ToolData, fanuc
      - l99.driver.fanuc.collectors.ProductionData, fanuc
      #- l99.driver.fanuc.collectors.ProductionDataExternalSubprogramDetails, fanuc
      #- l99.driver.fanuc.collectors.GCodeData, fanuc
      #- l99.driver.fanuc.collectors.AxisData, fanuc
      #- l99.driver.fanuc.collectors.SpindleData, fanuc
      #- l99.driver.fanuc.collectors.Macro, fanuc
      #- l99.driver.fanuc.collectors.Pmc, fanuc
```

ezioauditore0101, 2023-11-21:
Thanks for your response, I will try now and get back to you with report of this.

ezioauditore0101, 2023-11-21:
This time production has also not updated.
Please find CONFIG, LOG and MQTT screen capture. (Extension changed for uploading)
Config - [config.machines.txt](https://github.com/Ladder99/fanuc-driver/files/13428065/config.machines.txt)
Broker - [image]
Log file - [driver.2023-11-21.txt](https://github.com/Ladder99/fanuc-driver/files/13428063/driver.2023-11-21.txt)

MRIIOT, 2023-11-21:
It doesn't look like your logging is set to TRACE.  We should be seeing a lot more in the log.  Can you turn up all the loggers to TRACE?

ezioauditore0101, 2023-11-22:
I have made all from Debug to TRACE and generated a log; machine to refer is DMG. 
[driver.2023-11-22.txt](https://github.com/Ladder99/fanuc-driver/files/13433631/driver.2023-11-22.txt)
[nlog.config.txt](https://github.com/Ladder99/fanuc-driver/files/13433633/nlog.config.txt)
[image]

MRIIOT, 2023-11-22:
What is "MQTT IP" ?  

```
"l99.driver.fanuc.transports.MQTT, fanuc": {
        "topic": "fanuc/{{machine.Id}}/{{veneer.Name}}{{if veneer.SliceKey}}/{{veneer.SliceKey}}{{end}}",
        "net": {
          "type": "tcp",
          "ip": "MQTT IP",
          "port": 1883
        },
        "anonymous": false,
        "user": "MQTT_USER",
        "password": "MQTT_PASS"
      },
```

MRIIOT, 2023-11-22:
Everything looks fine, except the fact that we supposedly are connected to `tcp://MQTT IP:1883` broker, but I don't think '
MQTT IP' is valid.

```
2023/11/22 09:55:46.590|INFO|[DMG] Connecting broker 'tcp://MQTT IP:1883/DMG': MQTT IP:1883 |l99.driver.fanuc.transports.MQTT|
2023/11/22 09:55:46.641|INFO|[DMG] Connected broker 'tcp://MQTT IP:1883/DMG': MQTT IP:1883 |l99.driver.fanuc.transports.MQTT|
```

ezioauditore0101, 2023-11-22:
Actually yes, I updated my MQTT Broker IP to "MQTT IP" along with username and password.

ezioauditore0101, 2023-11-22:
Also, I am getting the FIRST ENTRY for SPINDLE 1 whenever the driver starts, which is referred in the screenshot I shared. You will see "fanuc/DMG/production/1" on an entry above where the driver started for very first time, while "fanuc/DMG/production/2"  is appearing frequently (on-change).

MRIIOT, 2023-11-22:
Ok, I think I know what you mean.  Which fanuc-driver Windows release are you using?  Can you share the DMG config again?

MRIIOT, 2023-11-22:
Sounds like this might be your issue https://github.com/Ladder99/fanuc-driver/issues/92

ezioauditore0101, 2023-11-27:
rebuilt with updated version. problem has been resolved.
thank you very much!

closed 2023-11-27 by ezioauditore0101