Ladder99/fanuc-driver #112: otel error error="<nil>" when running docker in WSL2 is it a Docker thing?

suttontools, 2024-07-16:
Hi,
I have successfully run the l99 stack on WSL2 and RPi but I am trying out the fanuc_driver with a real machine, again in WSL2.  I have used
`export COMPOSE_PROFILES=fanuc,mtc_2203,admin` and ` export FOCAS_TGT=linux64-f139dd5`. Prior to using admin and mtc_2203 I used fanuc,mtc only, and still got those now otel errors.  I assume they are due to using the --verbose command.  Now I haven't got the configuration files correct as I haven't got a pretty agent or the id of the machine in the XML but with the otel error I think I might be missing something?  The containers run, and the two (?non?) errors seem to be there even when bring the containers down

```
steve@HP-ELITEDESK:~/fanuc/fanuc-driver/examples$ docker compose --project-name fanuc --file compose.yml --verbose start
DEBU[0000] Enabled Docker Desktop integration (experimental) @ unix:///var/run/docker-cli.sock
[+] Running 5/5
 ✔ Container dozzle        Started                                                                                                                                           1.2s
 ✔ Container mosquitto     Started                                                                                                                                           1.4s
 ✔ Container fanuc_driver  Started                                                                                                                                           1.4s
 ✔ Container agent_2203    Started                                                                                                                                           1.1s
 ✔ Container portainer     Started                                                                                                                                           1.2s
DEBU[0001] otel error                                    error="<nil>"
DEBU[0001] otel error                                    error="<nil>"

docker compose --project-name fanuc --file compose.yml --verbose down
DEBU[0000] Enabled Docker Desktop integration (experimental) @ unix:///var/run/docker-cli.sock
[+] Running 3/5
[+] Running 6/6squitto     Removed                                                                                                                                           1.0s
 ✔ Container mosquitto     Removed                                                                                                                                           1.0s
 ✔ Container fanuc_driver  Removed                                                                                                                                          10.7s
 ✔ Container dozzle        Removed                                                                                                                                           0.7s
 ✔ Container portainer     Removed                                                                                                                                           0.8s
 ✔ Container agent_2203    Removed                                                                                                                                          10.6s
 ✔ Network ladder99        Removed                                                                                                                                           0.3s
DEBU[0011] otel error                                    error="<nil>"
DEBU[0011] otel error                                    error="<nil>"
steve@HP-ELITEDESK:~/fanuc/fanuc-driver/examples$
```

Thanks for your hard work

MRIIOT, 2025-03-28:
I have seen this as well but everything functions as expected.

closed 2025-03-28 by MRIIOT

ottobolyos, 2025-03-28:
I presume `otel` stands for Open TELemetry (collector), therefore, it could mean that either it is not configured (but something expects it to be configured) or it is misconfigured. :thinking: