Ladder99/fanuc-driver #114: Encountering issue when enabling ProductionDataExternalSubprogramDetails collector

karshatwork, 2024-09-03:
Hi,

I am trying to log data from "ProductionDataExternalSubprogramDetails" collector to SHDR. When I enable it in machines.yml, I start getting an error:
`Strategy sweep failed at segment Path Microsoft.CSharp.RuntimeBinder.RuntimeBinderException: Cannot apply indexing with [] to an expression of type 'int'`

traced to:
https://github.com/Ladder99/fanuc-driver/blob/f139dd5be82b41d0e914a6f9c049c0b14aa5ca85/fanuc/veneers/ProductionDataExternalSubprogramDetails.cs#L79

Can someone help me figure this out...

Nice225, 2024-12-12:
I have the same issue, can someone help.

MRIIOT, 2025-03-28:
That particular collector was for a specific use case when the programs are on the C drive instead of CNC mem.  Leave it disabled.