Ladder99/fanuc-driver #116: Macro variable unseen in generated devices.xml

smtpl-iiotian, 2024-11-14:
@MRIIOT  the generated devices.xml doesn't has the macro and pmc variables. I am able to see macro and pmc data being fetched in the trace log.
Please guide how and where to add macro variables in the generated devices.xml

closed 2024-11-15 by smtpl-iiotian

MRIIOT, 2025-03-28:
This section needs to be modified for the macros to come through devices.xml.  Or you can modify the devices.xml manually to include them.

https://github.com/Ladder99/fanuc-driver/blob/266991c03b3b5b62f511d3b26756148b63d967b4/examples/docker/config.system.yml#L141-L301

juhev, 2025-03-28:
@MRIIOT Thank you for your answer, by now I figured this out by myself but the agent still states the content of the macro as unavailable.  I changed the devices.xml, put in the macro-collector in the config.machines.yml and added the specific macro-variable at the bottom of that file just like you did in an example (i just changed the address to the one I want to read). Are there any more changes neccessary that I am not aware of?

juhev, 2025-03-31:
Now I can read the macro variables. For anyone else wondering here's a step-by-step instruction to do so:

1. In config.machines.yml, if not already present, add the paragraph     
```
  l99.driver.fanuc.collectors.Macro, fanuc:
      map:
        - id: test_macro
          path: 1
          address: !!int 512
```
The id is the name of the macro you want to read, path is usually 1 but can be 2 and address is the line of the macro (in my case 512)

2. As mentioned by MRIIOT above, you have to write the Macro variable you are looking to read in the config.systems.yml, so it can appear in the devices.xml
```
<DataItem category="SAMPLE" id="{{device}}_p{{path}}_test_macro" type="MACRO"/>
```

4. At the bottom of that file you can also find a paragraph called `l99.driver.fanuc.veneers.Macro, fanuc:`. This is where the data is linked. The macro-variable consists of address and value, so you have to specify what you want to read. Add the following line in order to do so: 
```
ShdrSample p+`_test_macro` data.test_macro.value;
```

These changes should make the content of the macro variable available. If not, set the log to trace and see if anything is read.
If I forgot anything please feel free to add.